EtchedPixels/FUZIX #65: non-deterministic builds with sdcc

geijoenr, 2015-01-11:
I am observing the same source might compile to a binary that crashes in different places during boot depending on max-allocs-per-node build option and on the placement of kprintf's (a build with debug enable in process runs further than one without debug enabled).

do you also see this?

EtchedPixels, 2015-01-11:
max-allocs-per-node tends to move things around in memory a bit as does kprintf. Some of the emulators have quite nice debuggers which can help as you can set watchpoints in the emulator or hack it to log any writes to allegedly 'code' space.

geijoenr, 2015-01-20:
indeed, it turns out I was overwritting some bytes in the '_initialized' segment; and the variables affected changed depending on the offset introduced by the additional code from the kprintf's (or from any other change, for that matter), producing hangs in different places.

EtchedPixels, 2015-01-22:
ouch..

geijoenr, 2015-01-30:
I think this can be closed.

closed 2015-01-30 by geijoenr