EtchedPixels/FUZIX #68: gettext support

erkinalp, 2015-01-14:
FUZIX currently supports English only. GNU gettext supports localisation using .po and .mo files. With gettext, we will get proper message translation.

EtchedPixels, 2015-01-14:
Technically we only support "C" locale (which is the classic C language locale with ascii symbols and ascii order sorting). English for example would sort AaBbCcDd ..  while 'C' sorts ABCDabcd.

GNU gettext is way way too big for anything that could run on an 8bit micro. In fact I'm not entirely sure that runtime internationalisation is doable. On the other hand fishing out all the N_ marked strings and being able to compile them for a language might be a bit more doable but still hard.

Not on the short term TODO list - stuff like networking to do first

erkinalp, 2015-01-25:
I have seen you added the gettext header, are you sure? I was working on an embedded capable localisation library for use in another GPL2 project last summer.

EtchedPixels, 2015-01-25:
If you can make internationalization work in the tiny space we have I'd be delighted to replace the stub code with something better, either at compile or runtime.

Alan

erkinalp, 2015-01-25:
It is designed for ReactOS, so I will need to do an API transformation into UNIX system.

closed 2016-03-25 by EtchedPixels