EtchedPixels/FUZIX #97: Coexistence

erkinalp, 2015-01-31:
On m68k platform, FUZIX and Linux can coexist. We will need either of those:
a)Chainloader which loads FUZIX
b)Boot manager to choose among Linux and FUZIX plus (a)
c)GRUB compatible kernel image

EtchedPixels, 2015-01-31:
ucLinux for Atari ST has been dead since 1999, there is no ucLinux for early Macintosh, ucLinux for Amiga 68000 died a few years later. None of them used grub or grub compatible images.

So basically there is zero overlap.

closed 2015-01-31 by EtchedPixels

erkinalp, 2015-02-01:
x86 family is backwards compatible way from 8086 till AMD64. Later on x86 port, we will tackle on this again. DOS, Linux, BSD and FUZIX may coexist on a 386+. This may remain closed for now.

EtchedPixels, 2015-02-02:
Also ELKS (Linux 8086), and the loader from that may be ideal