EtchedPixels/FUZIX #98: x86 memory model and FUZIX conventions on it

erkinalp, 2015-02-02:
I have seen 16-bit address type in 8086 test code, probably remnant of 8-bit code. As you know, x86 has four overlappable segment selectors. We should document our kernel code segment and free space. malloc()/mmap() needs to be reworked to return "far" pointers on 8086 because sometimes 8086 programs use more than 64K. What about other system calls and stack segment? Are there any plans for common stack allocator?

EtchedPixels, 2015-02-02:
Only plans are to do the same as Minix, PC/IX etc did on PC hardware - only allow 64Kcode/64Kdata per process. So DS=ES=SS, CS may or may not match.

willsowerbutts, 2015-02-02:
Agree, pointers are 16-bits, processes and the kernel have either 64K (shared code+data) or 128K (split code+data). 

Later PDP-11s can do split instruction/data address spaces. In principle I think the socz80 MMU could be extended to do the same by watching the M1 line to distinguish data and instruction fetches.

Far pointers -- ugh. Haven't had to deal with those for quite a few years!

EtchedPixels, 2015-02-03:
M1 is only asserted for the first byte of the instruction (so peripherals can spot RETI/RETN) that would make it pretty hard to do on a Z80 as the MMU would need an instruction decoder. It's actually probably doable on 6502 though - the Z80 decoder would be convoluted but the 6502 decode is quite simple.

The Z280 did have split I/D but that was handled internal to the CPU  I think.

We'll need "far" pointers (or fortunately just 'far copy') for 8086, and also for 6509.

erkinalp, 2015-02-04:
Will the only platform supported be IBM PC family(platform-ibmpc, not created at the time of writing this) or will other (e.g. with 16M RAM/embedded versions and clones) be supported for 8086/88 family?

EtchedPixels, 2015-02-04:
I don't see why it should need a PC class device. The limit for an 8086 is 1MB unless there is banked RAM and then it gets a bit weird. 286 can only do unbanked 16MB by using some of the freakier undocumented features (LOADALL) or by doing protected mode.

erkinalp, 2015-02-05:
Yes, it is so on a regular 8086, but there are embedded ISA compatible versions and clones with eight bit segment selector shift instead of usual four.

closed 2015-05-19 by EtchedPixels