EtchedPixels/FUZIX #105: Shell is unsuitable for scripting

erkinalp, 2015-02-10:
Current shell `/bin/ssh` is only usable for interactive usage, no script flow control implemented, such as:
* `if` / `then` / `else` / `fi`
* `case` / `esac`
* `for` / `do` / `done`
This makes us unable to use the toolbox approach on FUZIX we are used to on UNIX and the likes.

EtchedPixels, 2015-02-10:
The source for the original Bourne shell is now freed, but it need someone to port it from early K&R style C to ANSI C.

erkinalp, 2015-02-10:
It is not even K&R C. It is worse, Pascal-ish C.

erkinalp, 2015-02-10:
http://minnie.tuhs.org/cgi-bin/utree.pl?file=V7/usr/src/cmd/sh/cmd.c

EtchedPixels, 2015-02-10:
Yes its written in mock Algol 8)

erkinalp, 2015-02-12:
I am porting sash now. It has a bunch of builtins, such as gzip and tar. When finished stripping(yes, there are too many builtins), I am going to upload it.

Zardoz89, 2015-02-12:
Thompson shell on ANSI C : https://github.com/JNeitzel/v6shell/blob/master/sh6.c

erkinalp, 2015-02-12:
Its license is GPL-incompatible.

EtchedPixels, 2015-02-12:
4 clause BSD. I don't really have a problem with that for add on tools

erkinalp, 2015-02-12:
There is osh from v6 shell package, with the same licence. osh is 8-bit clean, where as v6shell is not. I have PRd in #106.

erkinalp, 2015-03-07:
Trying to port toybox shell and utilities now. There are too many Linux specific bits in it.

erkinalp, 2015-04-01:
@EtchedPixels: Are you satisfied with v7 Bourne shell scripting possibilities?

EtchedPixels, 2015-04-01:
Yeah I think so - just needs ansifying

EtchedPixels, 2015-05-16:
Bourne shell ported. Ok it's still slightly buggy and needs some tweaking but I think this one is closed

closed 2015-05-16 by EtchedPixels