EtchedPixels/FUZIX #107: Software maintainer's guide needed

erkinalp, 2015-02-11:
There is a little of documentations on FUZIX. More documentation will make fork and distribution maintainers' job easier.

Zardoz89, 2015-04-17:
Or a "How To compile FUZIX", could be helpful. I'm trying to compile FUZIX (including patching SDCC 2.4.3 with https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/patches/SDCC ), and I keep getting errors.

dylanmc, 2015-10-20:
I'm also getting errors - trying to build FUZIX (host MacOS) following these helpful instructions: https://sinrega.org/?p=218 but am currently getting linker errors - fcc isn't able to find the libraries, apparently. I've populated the /opt/fcc/{lib/bin/include} directories, but not having luck yet. @Zardoz89 were you able to build / run?

Zardoz89, 2015-10-20:
Nope. But I don't try it since April :(

EtchedPixels, 2015-10-20:
What linker errors do you get ?

dylanmc, 2015-10-20:
> What linker errors do you get?

```
$ make
../../Library/tools/fcc -O2  -c decomp16.c
decomp16.c:261: warning 196: pointer target lost const qualifier
../../Library/tools/fcc -O2   decomp16.rel -o decomp16

?ASlink-Warning-Undefined Global '___sdcc_call_hl' referenced by module 'atexit'

?ASlink-Warning-Undefined Global '_strlen' referenced by module 'decomp16'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'decomp16'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'stat'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'atexit'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fflush'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'lseek'
make: *** [decomp16] Error 1
```

I tried passing -L and -l directives to point at the libraries, but that made no difference. Here's the contents of /opt/fcc/lib:

```
$ ls /opt/fcc/lib
c.lib		crt0.rel	crt0nostdio.rel	syslib.lib
```

The library build step seemed to complete without errors - is there a way to check if they're well-formed?

EtchedPixels, 2015-10-20:
Ok that one I know about I think (its what happens when the weeding of the SDCC supplied library fails)

dylanmc, 2015-10-21:
I'd be happy to help troubleshoot the build process - what went wrong in my build to cause this?

EtchedPixels, 2015-10-21:
The SDCC library build tries to import bits of the SDCC library to use in the FUZIX C library build. If it doesn't find the SDCC library then it ends up building a library missing those functions. strlen() is what you'd expect, while the ___sdcc_ functions are compiler internal helpers it can't find.

EtchedPixels, 2015-10-21:
https://github.com/EtchedPixels/FUZIX/issues/252
is the underlying issue - fix in progress when I get time

closed 2016-03-25 by EtchedPixels