EtchedPixels/FUZIX #108: posix_spawn api

erkinalp, 2015-02-11:
POSIX 1:2008 has posix_spawn() as a means to create new processes from scratch without the copy overhead of fork(). It has a parameter list like exec(), but acts on a new process instead of manipulating current one. Spawning is less resource intensive than forking. So we should use spawn internally, however, continue providing fork to external users for compatibility.

EtchedPixels, 2015-02-12:
posix_spawn is hideously complicated, and not a substitute for fork() either. vfork|() may be the better option. Not clear yet and it's not a priority.

erkinalp, 2015-02-12:
Dangerous to emulate in userspace using vfork(), should be implemented in kernel space:
http://ewontfix.com/7/

EtchedPixels, 2015-02-13:
posix_spawn won't fit in kernel space so that point is somewhat moot. The ones I'm currently looking at are vfork() - which is certainly hard for portability but this is C lib code so fairly ok, or a saner spawn() interface such as that from OS/9.

erkinalp, 2015-02-14:
spawnve() like MS Windows and QNX?

EtchedPixels, 2015-02-19:
Something like that yes - and the OS/9 one is also similar. They have better APIs than the design by committee POSIX one.

erkinalp, 2015-02-20:
POSIX committee has likely been inspired by lower level spawn call of MS Windows-CreateProcess(). It has many parameters like posix_spawn(), mostly left in their defaults or given NULLs in general use.

EtchedPixels, 2015-05-19:
For the shell we definitely want vfork() if anything.

closed 2015-05-19 by EtchedPixels