EtchedPixels/FUZIX #124: What about LEAN as alternate filesystem?

timofonic, 2015-02-19:
Maybe there's need of a FAT alternative that isn't patent encumbered. What about using LEAN or a subset of it?

LEAN is a (recursive) backronym that stands for "Lean yet Effective Allocation and Naming" and is an alternative to the FAT filesystem used by the FreeDOS-32 project.

LEAN uses an extent-based allocation algorithm and is designed to use the little-endian byte ordering. There have been a total of 6 versions (at the time of writing) and the changelog for each release can be found in the specification (see link).

LEAN is a free, simple, portable, personal file system created to provide an alternative to the proprietary -and partly patented- Microsoft FAT file system. It is primarly intended for media exchange and for use with embedded devices, and to expand the file system functionality of the FreeDOS-32 operating system.

The word LEAN is a recursive backronym for Lean yet Effective Allocation and Naming, based on the ironical contrary of FAT, as an English word. The initial inspiration of the file system came from the Linux Second Extended file system (ext2), followed by original research and integration of comments and criticism. Although independently developed, there are some similarities with the QNX 4 file system.

LEAN aims to be the simplest file system providing the full set of most common and useful file system features. Any design choices have gone in the direction of the greatest simplicity rather than performance. LEAN is designed for use even on platforms with limited memory and CPU power.

 LEAN provides the following features (mostly according to the creators of the specification):

*  Supports Unicode long file names (in contrary to FAT long names, patented by Microsoft).
*   Suppports extended attributes (name-value metadata, embedded in files or forks).
*   Requires a small memory footprint and isn't too intensive on the CPU.
*   Fairly easy to implement and to understand.
*   Suitable for both large and small volumes.
*   Hard and symbolic links.
*   POSIX-compatible. 

A more detailed overview of the filesystem's actual capabilities:
*   512-byte allocation unit (conventional sectors)
*  A maximum volume size of 2^63 - 1 sectors
*  A maximum file size of 2^64 - 1 bytes
*  File names of up to 4068 bytes (case sensitive and in UTF-8 format)

http://wiki.osdev.org/LEAN
http://freedos-32.sourceforge.net/lean/specification.php

EtchedPixels, 2015-02-19:
Way too big - and it needs 64bit values, which aren't fun on a low end 32bit cpu let alone on a 6502. It's lean -- for a modern file system, its not lean by 8bit standards.

The constraints on the fs for Fuzix are very very tight

- It has to be able to operate with 5 x 512byte buffers worst case
- It has to be clean of 64bit values in any sane form
- It has to be efficient on small media (eg 180K floppy discs)
- It needs to occupy tiny amounts of code memory
- Inodes need to be small in memory (<= 64bytes). I'd like to get the current ones smaller
- We generally can't afford to use bitmaps (too expensive to scan on CPUs like the 6502)

closed 2015-02-19 by EtchedPixels

pulkomandy, 2015-02-19:
Here is one I wrote for fun: http://pulkomandy.tk/_/_Development/_Amelie,%20a%20filesystem%20for%208-bit%20computers
Probably not useable as is, but some ideas and design may be useful.

erkinalp, 2015-02-19:
What about LEANER, which is the same as LEAN, except:
* 32 bit inode and sector counts
* No metadata checksumming
* Single-band
* Filename up to 250 characters
Needs to be implemented, however.

EtchedPixels, 2015-02-19:
erkinalp: still has bit maps still too big. The v7 file system is *really* *really* compact codewise. More useful will be to pull in some of the 2.11BSD fixes as noted in the TODO.

erkinalp, 2015-02-20:
Amélie might be suitable for a /home filesystem, but not for root filesystem(It has seggregated user blocks).

timofonic, 2015-02-20:
WhatsApp about this?
https://www.nonattached.net/lanyfs/spec.php

http://forum.6502.org/viewtopic.php?f=2&t=2244&sid=3387472fed5b2e0f4d290df42fb8d89c&start=15

EtchedPixels, 2015-02-20:
What about it ? Unbalanced b-trees and an unproven design

The v7 style fs works. It's tiny and the flaws and workarounds are well understood. It would be a major shift to change it and I don't plan to.