EtchedPixels/FUZIX #149: DOS emulator for x86 target

erkinalp, 2015-03-21:
There is a CP/M emulator for Z80 target. DOS emulator for x86 target would be good idea. FreeDOS is native to 8086 and under GPLv2, thus, only needed piece remaining is a loader and TSR to access FUZIX services.

EtchedPixels, 2015-03-21:
For 8086 we support 64K split I/D. DOS requires physically addressed real mode. It makes no sense to support DOS code on an 8086.

closed 2015-03-21 by EtchedPixels

erkinalp, 2015-03-24:
What about other way round? FUZIX loader from FreeDOS with a patched kernel and init? DOS uses full real mode, so FUZIX will appear like a user program occupying two segments worth of memory.

Zardoz89, 2015-03-25:
Loadlin loads Linux from DOS, so why not something similar for Fuzix-86 ?

EtchedPixels, 2015-03-25:
erkin: no - if it gets ported to a PC x86 then fuzix ittself is multiple segment but each app is split I/D so one or two segments.

Most of the interesting x86 targets don't run DOS anyway - if you've got 640K x86 PC you can run ELKS.

erkinalp, 2015-03-25:
x86 targets with less than 128K RAM? DOS needed first 64K and memory mapped I/O plus memory mapped BIOS.

Zardoz89, 2015-03-25:
> x86 targets with less than 128K RAM? DOS needed first 64K and memory mapped I/O plus memory mapped BIOS.

The original IBM PC had 16KiB of RAM : http://oldcomputers.net/ibm5150.html & http://www.old-computers.com/museum/computer.asp?c=274
And perhaps there  was a few x86 computers with the same amount of RAM (bus s-100 with a 8086 or 8088 CPU board ?). In any case, they would be very rare found today one of these machines without any RAM expansion.

EtchedPixels, 2015-03-26:
PC class systems are covered by ELKS. Things like the Psion3 or some of the embedded 80C186 devices on the other hand don't deal with ELKS well in the smaller configurations.

Anyway for the forseeable future unless someone decides to go write a lot of code the 8086 is just a compilation test.

erkinalp, 2022-04-10:
> PC class systems are covered by ELKS.

FUZIX has real Unix code in it, ELKS does not.