EtchedPixels/FUZIX #155: Cowsay

erkinalp, 2015-03-29:
The game which displays speech bubbles, companion of fortune. A version written in C preferred.

EtchedPixels, 2015-03-29:
____________________________________ 
/ That would be silly, and anyway it \
\ needs perl                         /
 ------------------------------------ 
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

closed 2015-03-29 by EtchedPixels

erkinalp, 2015-03-29:
https://github.com/0xAether/ccowsay
This is in C.

erkinalp, 2015-03-29:
Ccowsay is in GPLv3, unsuitable for inclusion.

erkinalp, 2015-03-29:
https://github.com/shkschneider/c_unix/blob/master/cowsay.c
This is in GPLv2 as specified in its readme.

erkinalp, 2015-03-29:
PR made in #156.

erkinalp, 2015-03-29:
/ This is         \
    \ written in C. /
    
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

erkinalp, 2015-04-02:
Now in #157.