EtchedPixels/FUZIX #161: Feature: C128 support?

darkstar, 2015-04-12:
The C128 has 128k RAM and a Z80 CPU with 4MHz (on newer models with VIC-II disabled), backed by a 6502-compatible "co-processor" in Z80/CPM mode, so it should theoretically be possible to run FUZIX on it.

erkinalp, 2015-04-12:
Can you give more information about this cooperatory system?

darkstar, 2015-04-12:
Only what [Wikipedia says](http://en.wikipedia.org/wiki/Commodore_128#CP.2FM_mode), because even though I do own a C128 I cannot run CP/M on it as I lack the required floppy drive (C-1571) and the CP/M boot disk

> From the source code of the C128 CP/M implementation, it is clear that the engineers originally planned to make it possible to run CP/M in the "fast" mode as well, with the 40-column output turned off and the Z80 running at an effective 4 MHz; however, this feature did not correctly function on the first-generation C128 hardware.

and

>An unusual feature of the C128 among CP/M systems is that some of the low-level BIOS services are executed by the 8502 instead of the Z80. The latter transferred control to the 8502 after having placed the pertinent parameter values in designated memory locations. The Z80 then turns itself off, being awakened by the 8502 at completion of the BIOS routine, with status value(s) available in RAM for inspection. The reason for this is so that the Commodore development team would not have to rewrite the KERNAL routines for two completely different CPU instruction sets. The source code for the BIOS was available.

EtchedPixels, 2015-04-12:
The C128 is probably doable if someone contributes a port. However you are probably going to need some kind of sensible storage system (the Commodore serial floppy drives are incredibly slow and while the 1571 is better its still very very slow) or an REU or similar attached. It would also need someone very familiar with the C128 hardware as its quite different to a "normal" Z80 based system.

erkinalp, 2015-04-13:
Both processors have almost equal compute performance. We will therefore need special optimisation for this target.

EtchedPixels, 2015-04-13:
Not really. The Z80 is faster for 16 and 32bit work at a similar clock. It also has a 16bit stack pointer and avoids a pile of really painful problems the 6502 has. It might be entertaining to be able to run both 6502 and Z80 apps but you'd keep the kernel Z80 I think.

Either way it would really be up to who contributed the port.

closed 2016-03-25 by EtchedPixels