EtchedPixels/FUZIX #169: build-filesystem

EtchedPixels, 2015-04-26:
Need a script to build a minimalist file system. Some of our machines only have 40 track floppies.

EtchedPixels, 2015-05-10:
Now have a script for building a smaller file system but probably want to go smaller still

EtchedPixels, 2015-05-22:
Down to 140K.. probably about the limit we can push

closed 2015-05-22 by EtchedPixels

erkinalp, 2015-05-25:
Still too much. I expected it to fit  in a 20-track/side, 2-side, 8
sector/track, 256b/sector ie. 80k floppy diskette.
23 May 2015 02:39 tarihinde "EtchedPixels" <notifications@github.com> yazdı:
>
> probably about the limit we can push
>
> —
>

EtchedPixels, 2015-05-25:
Not a chance, not with any material number of commands. We can fit a small fs in 140K, that should be fine.  A 3"  SD disk is 180K, a 5.25" 80 track SS disk is 200K, an Apple II disk is 140K.

erkinalp, 2015-05-26:
Seems possible with my single-user PR applied along with DOS-like set of utilities(init,sh,mkdir,rmdir,mknod,chmod,chown,cp,mv,rm,ed)

EtchedPixels, 2015-05-26:
Having the user features in the kernel costs nothing - no patch is needed. You can just not bother supporting it in your userspace and run always as root.

Yes you could probably build a sort of micro-shell with minimal versions of the core commands in, even then I suspect it would be very tight to get all that into the space and fitting into memory as one or two binaries with links in the busybox style.