EtchedPixels/FUZIX #172: z80pack simulator images not being created correctly

darkstar, 2015-04-27:
I tried building FUZIX for zhe z80pack simulator by following [this guide](https://sinrega.org/?p=218)

However, it looks like the boot floppies are not built correctly:
```
root@kubuntu:/usr/src/FUZIX/Kernel/platform-z80pack# ./createdrives
Making filesystem on device drivea.cpm with isize 8 fsize 390.
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Making filesystem on device drivei.cpm with isize 64 fsize 8192.
Opening drivei.cpm sd0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Fuzix UCP version 1.1ac. Type ? for help.
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
i_open: bad disk inode
mkdir: mknod error 0
Fuzix UCP version 1.1ac. Type ? for help.
i_open: bad disk inode
mkdir: mknod error 0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
chmod cat to 755
chmod: can't stat file 'cat': 2
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
i_open: bad disk inode
Can't open destination file './../../Applications/util/cp'; error 0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
chmod df to 755
chmod: can't stat file 'df': 2
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
chmod cp to 755
chmod: can't stat file 'cp': 2
Opening drivea.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
Opening drivea.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
cd: error number 20
Fuzix UCP version 1.1ac. Type ? for help.
Fuzix UCP version 1.1ac. Type ? for help.
i_open: bad disk inode
mkdir: mknod error 0
cd: error number 2
i_open: bad disk inode
_mknod: error 0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
cd: error number 20
Opening drivea.cpm sd0
Opening drivea.cpm sd0
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
Opening drivei.cpm sd0
fd=3, dev_offset = 0
Opening drivei.cpm sd0
Opening drivei.cpm sd0
fd=3, dev_offset = 0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 2
chmod mkfs to 755
chmod: can't stat file 'mkfs': 2
cd: error number 2
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Opening drivea.cpm sd0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
i_open: bad disk inode
Can't open destination file './../../Applications/util/df'; error 0
Fuzix UCP version 1.1ac. Type ? for help.
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
Fuzix UCP version 1.1ac. Type ? for help.
Fuzix UCP version 1.1ac. Type ? for help.
Opening drivea.cpm sd0
chmod init to 755
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
Fuzix UCP version 1.1ac. Type ? for help.
cd: error number 20
i_open: bad disk inode
Can't open destination file './../../Applications/util/ln'; error 0
Opening drivei.cpm sd0
chmod fdisk to 755
fd=3, dev_offset = 0
...
Can't open destination file './../../Applications/util/ed'; error 0
Can't open destination file './../../Applications/util/patchcpm'; error 0
i_open: bad disk inode
chmod passwd to 755
Can't open destination file './../../Applications/util/sed'; error 0
chmod: can't stat file 'passwd': 2
chmod mkdir to 755
Can't open destination file './../../Applications/util/grep'; error 0
Can't open destination file './../../Applications/util/passwd'; error 0
chmod: can't stat file 'grep': 2
chmod: can't stat file 'mkdir': 2
45346+0 records in
45346+0 records out
45346 bytes (45 kB) copied, 0,127427 s, 356 kB/s
390+0 records in
390+0 records out
199680 bytes (200 kB) copied, 0,00152064 s, 131 MB/s
0+1 records in
0+1 records out
128 bytes (128 B) copied, 0,00019373 s, 661 kB/s
z80pack drives "drivea.cpm" and "drivei.cpm" successfully generated
```

Trying to run the images on the emulator gives the following:
```
root@kubuntu:/usr/src/z80pack-1.26/cpmsim# ./cpmsim

#######  #####    ###            #####    ###   #     #
     #  #     #  #   #          #     #    #    ##   ##
    #   #     # #     #         #          #    # # # #
   #     #####  #     #  #####   #####     #    #  #  #
  #     #     # #     #               #    #    #     #
 #      #     #  #   #          #     #    #    #     #
#######  #####    ###            #####    ###   #     #

Release 1.26, Copyright (C) 1987-2015 by Udo Munk

CPU speed is unlimited

Booting...

-
FUZIX version 0.1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2015 Alan Cox <alan@etchedpixels.co.uk>
Devboot
480kB total RAM, 416kB available to processes (15 processes max)
Enabling interrupts ... ok.
bootdev: 0
Mounting root fs (root_dev=0): OK
Starting /init

panic: no /init
System halted, bye.
```

The files in Applications/util get built correctly, in fact everything up to the ```createdrives```script runs without errors.

Any ideas what's going wrong? I tried  using ucp directly:
```
root@kubuntu:/usr/src/FUZIX/Kernel/platform-z80pack# ../../Standalone/ucp drivea.cpm
Opening drivea.cpm sd0
fd=3, dev_offset = 0
Fuzix UCP version 1.1ac. Type ? for help.
unix: ?
UCP commands:
?|help
exit|quit
dir|ls [path]
cd path
mkdir dirname
mknod name mode dev#
chmod mode path
[b]get sourcefile [destfile]
[b]put uzifile
type|cat filename
dump filename
rm path
rmdir dirname
df
mount dev# path
umount path
unix: ls
drwxrwxrwx   6     1         512 Apr 27  115    8:54  ./
drwxrwxrwx   6     1         512 Apr 27  115    8:54  ../
----------   1    46           0                      bin
drwxrwxrwx   2    44         512 Apr 27  115    8:54  root/
drwxrwxrwx   2    43         512 Apr 27  115    8:54  usr/
-rw-rw-rw-   1    42        7000 Apr 27  115    8:54  init
-rwxr-xr-x   1    41       14221 Apr 27  115    8:54  fdisk
brw-r--r--   1    40         256 Apr 27  115    8:54  hda
crw--w--w-   1    37        1026 Apr 27  115    8:54  zero
unix: cd bin
cd: error number 20
unix: rm bin
unix: mkdir bin
unix: cd bin
unix: bget ../../Applications/util/init
i_open: bad disk inode
Can't open destination file '../../Applications/util/init'; error 0
unix: ls
drwxrwxrwx   2    46         512 Apr 27  115    8:55  ./
drwxrwxrwx   7     1         512 Apr 27  115    8:55  ../
unix: cd ..
panic: inode freed.
root@kubuntu:/usr/src/FUZIX/Kernel/platform-z80pack#
```

EtchedPixels, 2015-04-27:
Known bug - we need a script to build a small floppy of basic tools only. Also ucp blows up when we run out of space.

It's issues 168 and 169

If you put just a kernel on floppy and the image on hard disk and boot with that as root fs it should work, ditto if you hand build a small file system or hack the script to only install useful bits needed for rescue/install not stuff like banner.

darkstar, 2015-04-27:
I don't really understand the error "Can't open destination file". Shouldn't that be the source file?

So the error is not on disk 1 but on disk 2? Can't I just make the disk 2 larger so that everything fits?

EtchedPixels, 2015-04-27:
It gets itself in a twist, so the errors once it fills up make no sense at all

EtchedPixels, 2016-03-25:
For z80pack for the moment I think the answer is to just build a 4MB hard disk image for drive I

The underlying bug has its own entry so closing this

closed 2016-03-25 by EtchedPixels