EtchedPixels/FUZIX #191: tormodvolden contributing via unusual way

erkinalp, 2015-05-24:
Tormod Volden does not use usual way of contributing on GitHub despite having a GitHub account, instead, he sends e-mail patches to Alan as far as I see. Is he a contract worker for FUZIX?

EtchedPixels, 2015-05-24:
No he just prefers to send patches via email. Tormod does the 6809 port and also designs hardware toys for the Dragon 32 boxes

closed 2015-05-24 by EtchedPixels

tormodvolden, 2015-05-26:
I am contracted by Dragon Data Ltd. They were a bit slow on that OS-9 thing, but they are not gonna miss the FUZIX train. Alan has promised a framebuffer with **lowercase** letters (!) so this time the educational market won't fail.

"git send-email" *is* the way to contribute to open-source projects. Real men and wise women don't rely on a proprietary web service to get their sh#t done. And I find all the "merge" commits in the log for single commits annoying.

erkinalp, 2015-05-26:
GitHub has its own command line tool called hub for this purpose. All repository operations(including creating pull requests and commenting) can be done using this tool and it uses proffessional GitHub API, not the web one.

erkinalp, 2015-05-26:
@tormodvolden: If you rebase commit to the upstream and merge using git's command line tool, you should not see that behavior.

EtchedPixels, 2015-05-26:
git send-email works perfectly well.

erkinalp, 2015-06-05:
> I am contracted by Dragon Data Ltd.

It was bankrupted long ago according to Wikipedia.

EtchedPixels, 2015-06-06:
#include <humour.h>

erkinalp, 2015-06-19:
Error: illegitimate include file "/usr/include/humour.h" which is not a regular file but is a device node