EtchedPixels/FUZIX #201: uptime on 6809 outputs the "\n" printed last, before anything else ???

EtchedPixels, 2015-06-06:
wtf ?

beretta42, 2015-06-17:
Hmmm... same exact thing happens on my coco3 port.  Here I thought it was my tty swallowing the final \n

beretta42, 2015-07-19:
gcc is optimizing the final printf("\n") into a putchar call.  The lib's putchar (see putchar_wrapper.c)  calls write directly, thereby short-cutting the lib's buffer.  and out pops the "\n" before the rest of the line.  I changed putchar:

int putchar(int ch)
{
	return fprintf(1,"%c", ch );
}

which eliminates the pre-output line "\n", but leaves the problem of no trailing "\n"...

EtchedPixels, 2015-07-19:
Yes it would  fprintf(stdout, ...) not 1

I think the actual problem is that the putc/getc macros are broken. They were of dubious worth anyway so I'll just replace them with unconditional use of fputc() and fgetc(). That ought to fix it.

beretta42, 2015-07-19:
yup.. I changed the Makefile.6809 to use regular "putchar.c", rather than
"putchar_wrapper.c", and it indeed seems to work.  I haven't been able to
prove it works on dragon or dragon-nx32, but it works on coco3.

hah... the "1" must have snuck in from "write(1,...)"  :)

closed 2015-07-31 by EtchedPixels