EtchedPixels/FUZIX #219: C11 stdio locked fopen() freopen() mode

erkinalp, 2015-07-06:
"x" mode should cause file to be opened locked exclusively. This also requires implementation of O_EXCL at Unix calls levels.

EtchedPixels, 2015-07-06:
While you can support "x" the C11 changed behaviour for "w" or "a" directly contradicts SVID and Unix standard behaviour.

O_EXCL is supported as is 3 argument open so all of these can be done but I'm dubious it is a good idea. Possibly it needs a separate -lc11 library one day. Certainly not a short term priority.

erkinalp, 2015-07-06:
> While you can support "x" the C11
> changed behaviour for "w" or "a"

Exclusive lock can be used for reading too(.i.e. "ex") Nobody else can read the file.  
This is default semantics with e.g. win32 and useful for network file servers.

EtchedPixels, 2015-07-06:
Its not supported in Unix at all, and a nightmare on networks so while x makes sense mandatory exclusive access locks are just not going to happen

erkinalp, 2015-07-06:
@EtchedPixels: There is no mandatory exclusive file lock on Windows either. Just comes by default, unlike Unix file open API, whilst shared locking is also available by calling relevant flag getting the file handle. What prevents an optional exclusive read lock aka "rx"?

EtchedPixels, 2015-07-07:
Windows implements mandatory locking - if you open a file for exclusive access nobody else can access it. That causes nightmares on networked systems and multi-user environments so the model used by flock() is advisory locking. Processes can choose to co-operate but anyone else with access to the file can choose not to bother.

erkinalp, 2015-07-08:
> Processes can choose to co-operate but
> anyone else with access to the file can
> choose not to bother.

What about file consistency in an exclusive locked file? Would partially modified files not cause issues on "I am the king, ignoring everyone else" side?

EtchedPixels, 2015-07-16:
Yes.. but that's not a problem the Unix OS tries to solve. An OS can't solve it because transactions are usually not single file based.

closed 2016-03-25 by EtchedPixels