EtchedPixels/FUZIX #225: Issue compiling Kernel under FreeBSD

gustauperez, 2015-08-17:
When compiling a kernel under FreeBSD AMD64/HEAD, platform msx2 (but also fails with z80pack) the build fails when expanding %.rel [1].

I first thought it was the make version in base interfiring, but creating an alias for make doesn't seem to solve the issue.

This happens with gnu make 4.1. Somehow it seems to have problems with the static pattern rules.

[1] [gus@portgus /usr/home/gus/FUZIX/Kernel]$ gmake V=1
     rm -f platform
     ln -sf platform-z80pack platform
     make -C platform-z80pack
     make[1]: don't know how to make %.rel:. Stop

EtchedPixels, 2015-08-17:
That sounds like your sub makes are not using GNU make. Are you sure your subshells and submake are inheriting the alias ?

gustauperez, 2015-08-18:
You're right, thank you. 

The problem happens with the direct invocation of make on Kernel/Makefile:112 because the aliases don't get inherit during the execution of make. Thus the make used is the BSD make (called make) instead the GNU make (which gets installed as /usr/local/bin/gmake)  

Creating a softlink to GNU make in my $HOME and playing with the $PATH variable (which gets inherit from the environment) allowed to proceed with the build.

Thank you.

EtchedPixels, 2015-08-21:
This really ought to work out of the box without hacks. As we use GNU syntax we should probably either replace all the submake calls with calls to $(MAKE) or use "gmake" so we always get the right make tool.

EtchedPixels, 2015-08-21:
I was thinking about replacing the entire mess with

echo $(which sdcc | sed -e "s/bin\/sdcc//")share/sdcc/lib/$CPU

which should work for any "normal" install of sdcc in any location.

gustauperez, 2015-08-21:
1.- As far I can tell, I'm not sure all Linux distros have gmake (Ubuntu simply has make). I'd use $(MAKE) with a default value like "make" and leave the user to possibility to change it to the right value (for example, I would  change it to gmake).

2.- About the sdcc detection, I'd agree with that solution, I think it should work with normal solutions where bin/ and share/sdcc share a common $PREFIX. Too bad sdcc doesn't provide its own .pc file to extract the right flags.

gustauperez, 2015-10-03:
I'd close this one.

closed 2015-10-03 by EtchedPixels