EtchedPixels/FUZIX #248: Segfault on ucp with FreeBSD

gustauperez, 2015-10-02:
This has been tested with FreeBSD 10.2 and FreeBSD 11 (HEAD) both on AMD64. When trying to use ucp, it segfaults.

I did build ucp with debug symbols. A simple execution of ucp (with no arguments) causes it to segfault [1]. Apparently this happens when calling printf at ucp:75, if I remove that printf the ucp exits gracefully. 

The bt is huge, it looks like an infinite loop somewhere. 

If I add a simple printf at ucp:main, the program also segfaults, with any string. 

With Standalone/[fsck|mkfs] (for example) eveything goes fine, printf gives no problem. 

[1] http://dpaste.com/0T4PYQ7

EtchedPixels, 2015-10-02:
I would bet some of the _open, _read and similar symbols ucp uses ti hide the Fuzix read/write etc clash with FreeBSD stdio internals and causes mayhem. Changing all the _foo to fuzix_foo will probably cure it

fsck and mkfs are ok as I cleaned them up not to use the hacked up kernel code used for ucp

EtchedPixels, 2015-10-02:
Try now

gustauperez, 2015-10-02:
Thank you. 

Now mknod defined at ucp:52 clashes [1] with the libc one. Perhaps that mknod should be renamed to ucp_mknod [2], I guess that the mknod at ucp:52 is internal to ucp.c. Perhaps the same should be applied to the other functions at ucp:[46-58] to prevent clash problems with the same symbols in the libc?
 
[1] http://dpaste.com/0FH3P5G
[2] http://dpaste.com/0F1W8G1

gustauperez, 2015-10-03:
The changes on commit 9911627 fixes ucp.

EtchedPixels, 2015-10-03:
Yay

closed 2015-10-03 by EtchedPixels