EtchedPixels/FUZIX #251: Problem building userspace

gustauperez, 2015-10-03:
According to the user space wiki, fcc and friends are needed. However the build of the syscall helper fails because of the syscall_name.h header is not in the same directory. Helping the compiler passing it the right place to search the header helped the build.

Perhaps it should be noted in the wiki or a simple Makefile may be deployed on Library/tools to build those.

[1] https://github.com/EtchedPixels/FUZIX/wiki/Building-User-Space
[2] http://dpaste.com/37BKFX7
[3] CFLAGS=-I../../Kernel/include make syscall

EtchedPixels, 2015-10-03:
Humm works for me because Library/Makefile adds the CFLAGS

gustauperez, 2015-10-03:
Then I guess the wiki should be updated to show that those are built from Library instead of cd Library/tools && make binman fcc fsize syscall.

EtchedPixels, 2015-10-03:
It says "Everything here is now hopefully historical except for the notes about where the tools need to go."

closed 2015-10-03 by EtchedPixels