EtchedPixels/FUZIX #253: Problem with clang and header files in Library/Makefile

gustauperez, 2015-10-03:
In Library/Makefile there is a header file listed as a requirement for some targets. Because the rules are implicit, gmake adds the header file to the compile command [1]. This breaks the compilation for clang [2].

Because syscall_name.h is not generated in the build, I think it could be safely removed from the list of requirements for those rules [3]. 

[1] cc -I../Kernel/include    tools/syscall_6502.c ../Kernel/include/syscall_name.h   -o tools/syscall_6502

[2] cc: error: cannot specify -o when generating multiple output files
<builtin>: recipe for target 'tools/syscall_6502' failed
gmake: *** [tools/syscall_6502] Error 1

[3] http://dpaste.com/2876HJK

EtchedPixels, 2015-10-08:
Fixed I think

gustauperez, 2015-10-08:
It does and I'd close the issue. Thank you.

closed 2015-10-08 by EtchedPixels