EtchedPixels/FUZIX #260: Clean alternatives to Unix memory hoggy interfaces

EtchedPixels, 2015-10-18:
Right now the obvious ones are

opendir (sucks in malloc)
termcap (has an ABI defined honking great buffer)

Probably we need an

opendir_r(const char *path, struct somethingopaque *)

where the caller provides the buffer

termcap is a bit harder

EtchedPixels, 2015-10-18:
Consider tgetent_s()

erkinalp, 2015-10-18:
opendir_r() 's second parameter is per mount, right?

EtchedPixels, 2015-10-21:
Possibly allow a build without tgetent() but instead tscanent() so it passes each entry to a provided function. That makes the parser very very small and low memory.

beretta42, 2015-12-24:
opendir_r shouldn't be too hard.  regular opendir() could then just call opendir_r() passing a static struct.   We'll probably need to add a flag to the _dir/DIR struct to indicate to closedir() whether it should try to free() the subject struct.  The DIR struct is quite big (512 bytes +), so maybe regular opendir() should still calloc() this memory?

EtchedPixels, 2015-12-24:
opendir_r is on my TODO list - it's mostly a case of restructuring thing so that sizeof(DIR) produces the right answer and you can then do

DIR p
opendir_r(".", &p)

or similar

EtchedPixels, 2015-12-29:
opendir_r/closedir_r: DONE

EtchedPixels, 2016-09-01:
termcap turns out to be a bitch to rework into something very compact. Maybe terminfo is actually a better thing to support ?

In the mean time implement a helper app and test it with ubasic.

closed 2016-09-01 by EtchedPixels