EtchedPixels/FUZIX #275: inittab

DimkaM, 2015-10-25:
Tell me, please, where to get a working inittab?

EtchedPixels, 2015-10-25:
Standalone/filesystem-src/etc-files/inittab

DimkaM, 2015-10-25:
Thank you.
So the problem is not in the script.
I'll catch the bugs in my code.
"FUZIX version 0.1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2015 Alan Cox <alan@etchedpixels.co.uk>
Devboot
4096kB total RAM, 3968kB available to processes (15 processes max)
Enabling interrupts ... ok.
 SD drive 0: hda:
Mounting root fs (root_dev=0): OK
Starting /init
init version 0.9.0ac#1
: is not an identifier"

beretta42, 2015-10-25:
4 Megs of RAM?  cool!  which platform/machine has that?  MSX2?

DimkaM, 2015-10-25:
http://nedopc.com/zxevo/zxevo_eng.php

EtchedPixels, 2015-10-25:
one thing to check is to delete inittab. init should then just load the shell up. That may help find the bug.

DimkaM, 2015-10-25:
По вашему совету, удалил inittab.
Ошибка осталась.
Буду копать, где проблема.

Извините, я использую googletranslate

DimkaM, 2015-10-25:
On your advice, I removed inittab.
The error remained.
I will dig, where the problem is.

Sorry, I use googletranslate

DimkaM, 2015-10-25:
I am a disabled driver a ps/2 keyboard. An error has disappeared.
The problem in my keyboard driver.
I will try to enter characters via the rs232.

closed 2015-10-25 by DimkaM