EtchedPixels/FUZIX #283: Finish up SIGCONT, SIGSTOP handling

EtchedPixels, 2015-11-14:


EtchedPixels, 2015-12-26:
Logic put in place for the basics (SIGSTOP, SIGCONT) without TSTP or job control. Needs testing done

beretta42, 2016-02-25:
Initial test of SIGSTOP panic()'s on an infinite loop:

_chksigs() calls psleep() on pending SIGSTOP
psleep() calls switchout()
switchout() calls _chksigs()  (and starts the infinite loop)

I temporarily disabled the call to _chksig() in my tricks.s (coco3)  and SIGSTOP and SIGCONT seem to behave better.

EtchedPixels, 2016-02-25:
On Thu, 25 Feb 2016 08:53:35 -0800
"Brett M. Gordon" <notifications@github.com> wrote:

> Initial test of SIGSTOP panic()'s on an infinite loop:
> 
> _chksigs() calls psleep() on pending SIGSTOP
> psleep() calls switchout()
> switchout() calls _chksigs()  (and starts the infinite loop)

Ick.. hacky change pushed

beretta42, 2016-02-26:
It is still broke.  psleep() sets it's process to P_SLEEP before (re)calling _chksig.  :(

https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/process.c#L39

EtchedPixels, 2016-02-27:
Sigh. I'll have a look at exactly what is supposed to happen first, then worry about a correct fix.

beretta42, 2016-02-28:
I'm in no rush.  I'll keep on testing the newer level two stuff.

EtchedPixels, 2016-08-31:
Had another quick go at this in my break

closed 2016-08-31 by EtchedPixels