EtchedPixels/FUZIX #297: hardware sockets

DimkaM, 2015-11-15:
I have a network card ZXUSBNET, built on the base of chip wiznet w5300(http://www.nedopc.com/zxnetusb/zxnetusb.php).
Are there any plans in fuzix-kernel, hardware support sockets tcp/ip and udp/ip?
Or the need to implement support in a particular platform?

Sorry, I do not speak english well.

beretta42, 2015-11-15:
Alan has adding very preliminary coding for tcp/ip in fuzix... mostly
defines and the such.   I am sure its planned, but seems to be far down the
road.

by the way:  that a cool board you linked to!  I cant wait till we have
more 8 bitters on the internet.

-B

EtchedPixels, 2015-11-15:
I've started sketching out tcp/ip. I really want to get graphics working nicely first, fix the bugs in the basic support and tidy up the main supported platforms.

I have written (but not tested) the system call code for the networking. I would like to support hardware based tcp/ip (like the Wiznet) as well as software TCP/IP, but it's important they all get the same BSD socket API so applications can't tell what hardware/software is in use.

erkinalp, 2015-11-15:
> I cant wait till we have more 8 bitters on the internet.
Though, they will see garbage if they visit github in a text-only browser. ssh, yes. www, maybe.

DimkaM, 2015-11-16:
Thanks to all.

closed 2015-11-16 by DimkaM