EtchedPixels/FUZIX #312: Level 3: Add pledge()

erkinalp, 2015-12-27:
Capability dropping is a way to secure programs. It is based on whitelisting system calls by pledge per process or per thread. All are whitelisted by default.

EtchedPixels, 2015-12-27:
Wrong OS. I'd be surprised if Fuzix ever ended up running on hardware with full MMU protection, it's certainly not designed to be a formally secure OS.

closed 2015-12-27 by EtchedPixels

erkinalp, 2015-12-27:
It is needed for NetBSD-FUZIX source code compatibility.

erkinalp, 2015-12-27:
It is needed for source code compatibility with OpenBSD.

EtchedPixels, 2015-12-27:
Why would we care. If you want OpenBSD run OpenBSD.

erkinalp, 2015-12-27:
I have seen you were adding 4.4BSD system calls, it will/would eventually lead to adding current BSD variants.

EtchedPixels, 2015-12-27:
I'm adding a small set of calls needed to implement job control on bigger boxes. I've got zero interest in adding all the broken crap in modern "Unix", job control is icky enough but unfortunately we seem to be stuck with it.

erkinalp, 2015-12-27:
> the broken crap in modern "Unix"

MAC is a need even in these small boxes. pledge() is the simplest MAC framework around with more than two priviledge levels, depends on no other calls.

EtchedPixels, 2015-12-27:
Please get a clue

- Fuzix runs on machines that don't even have memory protection so what protections it has are for user convenience only (so you don't accidentally delete the wrong files)
- If you want to write a secure OS you'd start from scratch and write it in something like GO on proper hardware with a mathematically provable security model