EtchedPixels/FUZIX #333: Fails to build from scratch on my box while trying to document build process

ArielleA, 2016-03-18:
Hi,

I am trying to write a new build document so that it is easier for people to start with FUZIX. I have tried to use the formula presented here but am getting issues with building the libraries specifically when it comes to link the curses components as they don't assemble correctly. 
I'm on debian testing armhf, with SDCC build from yesterday, and am current with git.

Thankyou

Arielle :)

--

arielle@asuscb:~/src/FUZIX/Library/libs$ export PATH=$PATH:/opt/fcc/bin
arielle@asuscb:~/src/FUZIX/Library/libs$ make
fcc -O2 -c curses/charpick.c -o ./curses/
fcc -O2 -c curses/curs_set.c -o ./curses/
fcc -O2 -c curses/cursesio.c -o ./curses/
fcc -O2 -c curses/doupdt.c -o ./curses/
fcc -O2 -c curses/endwin.c -o ./curses/
fcc -O2 -c curses/initscr.c -o ./curses/
fcc -O2 -c curses/move.c -o ./curses/
fcc -O2 -c curses/mvcursor.c -o ./curses/
curses/mvcursor.c:15: warning 85: in function mvcur unreferenced function argument : 'oldx'
curses/mvcursor.c:15: warning 85: in function mvcur unreferenced function argument : 'oldy'
fcc -O2 -c curses/newwin.c -o ./curses/
fcc -O2 -c curses/options.c -o ./curses/
curses/options.c:14: warning 85: in function idlok unreferenced function argument : 'win'
curses/options.c:14: warning 85: in function idlok unreferenced function argument : 'flag'
curses/options.c:62: warning 85: in function keypad unreferenced function argument : 'win'
curses/options.c:62: warning 85: in function keypad unreferenced function argument : 'flag'
curses/options.c:71: warning 85: in function meta unreferenced function argument : 'win'
curses/options.c:71: warning 85: in function meta unreferenced function argument : 'flag'
fcc -O2 -c curses/overlay.c -o ./curses/
fcc -O2 -c curses/prnt.c -o ./curses/
fcc -O2 -c curses/prntscan.c -o ./curses/
fcc -O2 -c curses/refresh.c -o ./curses/
fcc -O2 -c curses/scan.c -o ./curses/
fcc -O2 -c curses/scrreg.c -o ./curses/
fcc -O2 -c curses/setterm.c -o ./curses/
fcc -O2 -c curses/tabsize.c -o ./curses/
fcc -O2 -c curses/unctrl.c -o ./curses/
fcc -O2 -c curses/waddch.c -o ./curses/
fcc -O2 -c curses/waddstr.c -o ./curses/
fcc -O2 -c curses/wbox.c -o ./curses/
fcc -O2 -c curses/wclear.c -o ./curses/
fcc -O2 -c curses/wclrtobot.c -o ./curses/
curses/wclrtobot.c:28: warning 84: 'auto' variable 'maxx' may be used before initialization
fcc -O2 -c curses/wclrtoeol.c -o ./curses/
fcc -O2 -c curses/wdelch.c -o ./curses/
fcc -O2 -c curses/wdeleteln.c -o ./curses/
fcc -O2 -c curses/werase.c -o ./curses/
fcc -O2 -c curses/wgetch.c -o ./curses/
fcc -O2 -c curses/wgetstr.c -o ./curses/
fcc -O2 -c curses/windel.c -o ./curses/
fcc -O2 -c curses/winmove.c -o ./curses/
fcc -O2 -c curses/winsch.c -o ./curses/
fcc -O2 -c curses/winscrol.c -o ./curses/
fcc -O2 -c curses/winsertln.c -o ./curses/
fcc -O2 -c curses/wintouch.c -o ./curses/
ls enter.rel __argv.rel abort.rel asctime.rel assert.rel atexit.rel atoi_small.rel bcmp.rel bcopy.rel bsearch.rel bzero.rel calloc.rel cfree.rel clock.rel closedir.rel closedir_r.rel clock_gettime.rel clock_getres.rel clock_settime.rel creat.rel crypt.rel ctermid.rel ctime.rel cuserid.rel difftime.rel err.rel errno.rel error.rel execl.rel execv.rel execvp.rel exit.rel fclose.rel fflush.rel fgetc.rel fgetgrent.rel fgetpwent.rel fgetpos.rel fgets.rel fopen.rel fork.rel fprintf.rel fputc.rel fputs.rel fread.rel free.rel fsetpos.rel ftell.rel fwrite.rel getcwd.rel getenv.rel __getgrent.rel getgrgid.rel getgrnam.rel getloadavg.rel getlogin.rel getopt.rel getpw.rel __getpwent.rel getpwnam.rel getpwuid.rel gets.rel gettimeofday.rel getw.rel gmtime.rel gmtime_r.rel grent.rel inet_addr.rel inet_aton.rel inet_network.rel inet_ntoa.rel inet_ntop.rel inet_pton.rel index.rel isatty.rel killpg.rel libintl.rel localtim.rel localtim_r.rel lseek.rel lsearch.rel lstat.rel ltoa.rel ltostr.rel malloc.rel mkfifo.rel mkstemps.rel nanosleep.rel opendir.rel opendir_r.rel pause.rel perror.rel popen.rel printf.rel putenv.rel putgetch.rel putpwent.rel putw.rel pwent.rel qsort.rel raise.rel rand.rel readdir.rel readlink.rel realloc.rel regerror.rel regsub.rel remove.rel rewind.rel rewinddir.rel rindex.rel seekdir.rel setbuffer.rel setenv.rel setlocale.rel setvbuf.rel settimeofday.rel sleep.rel sprintf.rel stat.rel stdio0.rel stime.rel strcasecmp.rel strcasestr.rel strdup.rel stricmp.rel strlcpy.rel strncasecmp.rel strnlen.rel strnicmp.rel strsep.rel strxfrm.rel strcoll.rel strsignal.rel strtod.rel strtol.rel swab.rel system.rel telldir.rel time.rel tmpfile.rel tmpnam.rel ttyname.rel tzset.rel ungetc.rel usleep.rel utent.rel utimes.rel utsname.rel vfprintf.rel vprintf.rel wait.rel xitoa.rel pathconf.rel gethostname.rel sysconf.rel confstr.rel memccpy.rel getpass.rel isalnum.rel isalpha.rel isascii.rel isblank.rel iscntrl.rel isdigit.rel isgraph.rel islower.rel isprint.rel ispunct.rel isspace.rel isupper.rel isxdigit.rel toupper.rel tolower.rel tcgetattr.rel tcsetattr.rel tcdrain.rel tcflow.rel tcflush.rel cfmakeraw.rel cfspeed.rel revoke.rel fscanf.rel scanf.rel sscanf.rel vfscanf.rel vscanf.rel vsscanf.rel termcap.rel curses/charpick.rel curses/curs_set.rel curses/cursesio.rel curses/doupdt.rel curses/endwin.rel curses/initscr.rel curses/move.rel curses/mvcursor.rel curses/newwin.rel curses/options.rel curses/overlay.rel curses/prnt.rel curses/prntscan.rel curses/refresh.rel curses/scan.rel curses/scrreg.rel curses/setterm.rel curses/tabsize.rel curses/unctrl.rel curses/waddch.rel curses/waddstr.rel curses/wbox.rel curses/wclear.rel curses/wclrtobot.rel curses/wclrtoeol.rel curses/wdelch.rel curses/wdeleteln.rel curses/werase.rel curses/wgetch.rel curses/wgetstr.rel curses/windel.rel curses/winmove.rel curses/winsch.rel curses/winscrol.rel curses/winsertln.rel curses/wintouch.rel regexp.rel > libc.l
ls: cannot access 'curses/charpick.rel': No such file or directory
ls: cannot access 'curses/curs_set.rel': No such file or directory
ls: cannot access 'curses/cursesio.rel': No such file or directory
ls: cannot access 'curses/doupdt.rel': No such file or directory
ls: cannot access 'curses/endwin.rel': No such file or directory
ls: cannot access 'curses/initscr.rel': No such file or directory
ls: cannot access 'curses/move.rel': No such file or directory
ls: cannot access 'curses/mvcursor.rel': No such file or directory
ls: cannot access 'curses/newwin.rel': No such file or directory
ls: cannot access 'curses/options.rel': No such file or directory
ls: cannot access 'curses/overlay.rel': No such file or directory
ls: cannot access 'curses/prnt.rel': No such file or directory
ls: cannot access 'curses/prntscan.rel': No such file or directory
ls: cannot access 'curses/refresh.rel': No such file or directory
ls: cannot access 'curses/scan.rel': No such file or directory
ls: cannot access 'curses/scrreg.rel': No such file or directory
ls: cannot access 'curses/setterm.rel': No such file or directory
ls: cannot access 'curses/tabsize.rel': No such file or directory
ls: cannot access 'curses/unctrl.rel': No such file or directory
ls: cannot access 'curses/waddch.rel': No such file or directory
ls: cannot access 'curses/waddstr.rel': No such file or directory
ls: cannot access 'curses/wbox.rel': No such file or directory
ls: cannot access 'curses/wclear.rel': No such file or directory
ls: cannot access 'curses/wclrtobot.rel': No such file or directory
ls: cannot access 'curses/wclrtoeol.rel': No such file or directory
ls: cannot access 'curses/wdelch.rel': No such file or directory
ls: cannot access 'curses/wdeleteln.rel': No such file or directory
ls: cannot access 'curses/werase.rel': No such file or directory
ls: cannot access 'curses/wgetch.rel': No such file or directory
ls: cannot access 'curses/wgetstr.rel': No such file or directory
ls: cannot access 'curses/windel.rel': No such file or directory
ls: cannot access 'curses/winmove.rel': No such file or directory
ls: cannot access 'curses/winsch.rel': No such file or directory
ls: cannot access 'curses/winscrol.rel': No such file or directory
ls: cannot access 'curses/winsertln.rel': No such file or directory
ls: cannot access 'curses/wintouch.rel': No such file or directory
Makefile:85: recipe for target 'libc.l' failed
make: *** [libc.l] Error 2
arielle@asuscb:~/src/FUZIX/Library/libs$

beretta42, 2016-03-18:
I am far from good at the build process myself... but from above, it looks like the .rel files aren't being produced.  What's your makefile look like?  All those fcc command lines should probably specify a .rel for an output file, not the curses directory path:

fcc -O2 -c curses/charpick.c -o ./curses/
--should be --
fcc -O2 -c curses/charpick.c -o ./curses/charpick.rel

I'm a gcc 6809 man... so I could be way off base here....

EtchedPixels, 2016-03-20:
Makefile bug - happened to work because of old behaviour in fcc I think. Will push a fix shortly

closed 2016-03-20 by EtchedPixels