EtchedPixels/FUZIX #338: 6809: /var/run/utmp growing way too big / bad format

beretta42, 2016-04-07:
_lseek system call was being called with and offset of 0x0000 ffce  rather than 0xffff ffce.  Turns out just negating the sizeof() breaks gcc.  Here's a diff that make utmp work as advertised:

--- a/Library/libs/utent.c
+++ b/Library/libs/utent.c
@@ -117,7 +117,7 @@ pututline(const struct utmp * utmp_entry)
 #endif
 
   if ((ut=getutid(utmp_entry))!=NULL)
-      lseek(ut_fd, (off_t) -sizeof(struct utmp), SEEK_CUR);
+      lseek(ut_fd, (off_t) sizeof(struct utmp)*-1, SEEK_CUR);
   else if( ut_fd==-1 )
       return NULL;
   else
@@ -133,7 +133,7 @@ pututline(const struct utmp * utmp_entry)
     return NULL;
 
   /* I think this final lseek gets the result Nat was after ... RdB */
-  lseek(ut_fd, (off_t) -sizeof(struct utmp), SEEK_CUR);
+  lseek(ut_fd, (off_t) sizeof(struct utmp)*-1, SEEK_CUR);
 
   /* Ignoring untrapped errors */

beretta42, 2016-04-07:
A call like this seems to work also:

lseek(ut_fd, -((off_t) sizeof(struct utmp)), SEEK_CUR);

if this is any clearer.

EtchedPixels, 2016-04-09:
Fixed

closed 2016-04-09 by EtchedPixels

erkinalp, 2016-04-10:
> Turns out just negating the sizeof() breaks gcc.

Z80 and 6502 are 2s complement. You can use `~sizeof(...)+1`.

EtchedPixels, 2016-04-11:
No you can't. That still won't work, as well as being obscure. Think about the casting and type rules