EtchedPixels/FUZIX #341: newb questions for an attempted 6809 port

nealcrook, 2016-04-17:
Hi,

this weekend I have been playing around with the FUZIX code tree with a view to doing a port to a 6809 Multicomp system. I'm confused about what the memory map looks like (kernel vs applications) and would appreciate some feedback to the following questions. 

thanks

Neal.

1. Ignoring paging for now, my target is a 6809 with 64K of RAM except for
0xFFD0-0xFFFF which contains I/O and hw vectors.

I'm looking at dragon-nx32 for reference (and I can build this OK after a bit
of fumbling). dragon-nx32 has:

#define PROGBASE    0x8000  /* also data base */
#define PROGLOAD    0x8000  /* also data base */
#define PROGTOP     0xFC00  /* Top of program */

and fuzix.link looks like this:

section .vectors load 0x0100
section .start load 0x1c00
section .common
section .text
section .buffers
section .discard
section .udata load 0x7D00
section .text2 load 0x8010
section .bss high 0xFC00
section .data
section .videodata
section .video
entry start

I understand the 0xFC00 in fuzix.link -- allocating data from PROGTOP
downwards.

I understand that the .discard section contains stuff that will only get
used once, so this memory will get reclaimed.

I'm confused about references (in PORTING) to "the udata area" and "common
memory" - what do these terms mean?

I'm confused about where the kernel sits in the address space. Assuming it's
possible to have a system with no bank switching and no swapping, I thought
that an application would load at PROGBASE and that, therefore, the kernel
must be permanently memory-resident elsewhere in the address space..

but the .link file about is putting .text2 just above PROGBASE so I must
be wrong about something here?

For my target, where I have no constraints on memory map, I think I should set
PROGBASE to 0x100 and PROGTOP to 0xFFD0. Applications should be linked to (load
and) execute at 0x100.

If that's correct, please can you give some guidance on that my fuzix.link
should look like (ie, where I should put the kernel)

2. I read in PORTING that the limit on blocks is 65536 (32MBytes)
and that block 0 is reserved "as a boot block".
In Standalone/filesystem-src/build-filesystem is this text:

echo "For a 32MB (or larger) hard disk, use isize=256, bsize=65535"

is that figure of 65535 correct, or should it be 65536 (so that
the .dsk image is exactly 32MBytes instead of being 1 block short)

3. In config.h what is the difference between PROGBASE and PROGLOAD?

4. Assuming I get my port working and make a push request, am I correct to infer
your preferred indent style as: 1 hard tab per indent?

5. is there a better place to ask questions?

EtchedPixels, 2016-04-17:
The memory map is mostly up to you. The Dragon port has banks in the upper 32K and the lower 32K common so puts the userspace at 32K. We've not bothered with a relocating loader yet, although it's on my Z80 list.

The only thing that really matters is that your common memory is always mapped by the kernel whatever memory is currently paged in, your udata is paged with the process it belongs to (or in some cases copied), and your interrupt and syscall entry points do the right thing to end up in the common code entry points.

If you have paging I would follow the COCO3

Best place to discuss is probably the fuzix-devel google groups list.

EtchedPixels, 2016-04-17:
Oha nd for your questions

1a. 65535 -  or some values wrap

1b. PROGBASE - start of program memory
      PROGLOAD - where it loads

So for 6809 you might have PROGBASE at 0x0000 but PROGLOAD at 0x100 because of the direct page - only of course you can move it.

2. For the core code 1 tab yes - and slowly moving all the code to comply. For platforms, you own the platform so within reason your style.

tormodvolden, 2016-04-17:
Just for demystifying, IIRC, the only reason I put .text2 at $8010 and not at $8000 is that the dragon-nx32 kernel startup code probes for banks by writing to them at $8000 so any code there would be clobbered (could probably be solved by rewriting the original contents after probing - it was written before we had .discard being reclaimed and a more complicated probe would have eaten more than 16 bytes...), and I rounded up to $8010. In the dragon-nx32 family I have extension boards with 1, 4 or 16 banks (32KB each).

If you only have 64KB RAM but ROM can be paged in, you would be more in the same boat as the unfinished "coco2" port and the dropped "dragon" port.

nealcrook, 2016-04-17:
Alan, Tormod, thanks for the replies, I think that I can push ahead with a bootstrap loader now and see what happens.

>>If you only have 64KB RAM 
well, I actually have 1MByte and a coco-style 8KByte-block pager, but I thought it would be easier to get started with a simple system. Maybe that's wrong..

Neal.

nealcrook, 2016-04-18:
>>  the fuzix-devel google groups list.

Is this a private group? I searched for it, including explicitly here:

https://groups.google.com/forum/#!overview

without success

Neal.

EtchedPixels, 2016-04-18:
It's certainly simpler to start by treating your memory as say 56K banks with an 8K common at the top and then worry about block paging. In fact with 1MB of RAM you'd need to be running 15 things to need to either think about smaller allocations or swap out. For a 1-2MHz processor that's probably more than sufficient !

EtchedPixels, 2016-04-18:
On Mon, 18 Apr 2016 05:15:22 -0700
Neal Crook <notifications@github.com> wrote:

> >>  the fuzix-devel google groups list.  
> 
> Is this a private group? I searched for it, including explicitly here:
> 
> https://groups.google.com/forum/#!overview
> 
> without success
> 

Fixed.. apparently Google defaults to private.

Alan

erkinalp, 2016-04-18:
Still private.

EtchedPixels, 2016-04-18:
It's markd public - but I think it takes a bit of time for Google to update the indexes

closed 2016-04-18 by EtchedPixels

aralbrec, 2016-04-20:
Is the group intended to be members only?

EtchedPixels, 2016-04-21:
to stop spambots joining all the time yes