EtchedPixels/FUZIX #342: opendir_r & closedir_r

DimkaM, 2016-04-18:
need add 'opendir_r.c' & 'closedir_r.c' into Library/build.mk

EtchedPixels, 2016-04-18:
Library/build.mk may be missing a few oddments as I'm generally building using the Makefiles not the parallel build rules.

DimkaM, 2016-04-18:
```
diff --git a/Library/build.mk b/Library/build.mk
index 1022ade..e945aec 100755
--- a/Library/build.mk
+++ b/Library/build.mk
@@ -25,6 +25,7 @@ libc-functions.localsrcs += \
        clock_gettime.c \
        clock_settime.c \
        closedir.c \
+       closedir_r.c \
        confstr.c \
        creat.c \
        crypt.c \
@@ -102,6 +103,7 @@ libc-functions.localsrcs += \
        mkstemps.c \
        nanosleep.c \
        opendir.c \
+       opendir_r.c \
        pathconf.c \
        pause.c \
        perror.c \

```

DimkaM, 2016-04-18:
[patch.txt](https://github.com/EtchedPixels/FUZIX/files/224125/patch.txt)

closed 2016-04-21 by EtchedPixels