EtchedPixels/FUZIX #344: CONFIG_IDE vs DEVICE_IDE

nealcrook, 2016-04-23:
The kernel code uses a mixture of CONFIG_IDE and DEVICE_IDE. Is there a subtle difference between them (if so what?) or should they be unified? (if so, which is preferred?)

Likewise CONFIG_SDC and DEVICE_SD

thanks,

Neal.

beretta42, 2016-04-26:
Looks like the CoCo/Dragon platforms use DEVICE_IDE, and the plus3 and zx128 platforms use CONFIG_IDE for the same purpose.  They can probably be all combined into one define.

CONFIG_SDC and DEVICE_SD are very similar meaning too. But they are used to represent two different devices: a DEVICE_SD is for MMIO SD, and CONFIG_SDC is for CoCoSDC.  Not sure if these should be combined.

tormodvolden, 2016-04-27:
If they are not to be combined, CONFIG_SDC should probably be renamed to CONFIG_COCOSDC to reduce confusion, even if it is a bit longer.

nealcrook, 2016-05-20:
Here is my analysis and proposed clean-up. Please let me know of any objections. Otherwise, I'll roll these changes into a janitorial branch I have brewing.
````
$ grep -R DEVICE_IDE *
Kernel/dev/devide.h:   Define DEVICE_IDE if IDE hardware is present on your platform.
Kernel/platform-n8vem-mark4/config.h:#define DEVICE_IDE
Kernel/platform-coco3/config.h:#define DEVICE_IDE        /* enable if IDE interface present */
Kernel/platform-p112/config.h:#define DEVICE_IDE                  /* enable if IDE interface present */
Kernel/platform/README:* BUG other platforms have mixture of CONFIG_IDE and DEVICE_IDE and CONFIG_SDC and DEVICE_SD
Kernel/platform-dragon-nx32/config.h:#define DEVICE_IDE              /* enable if IDE interface present */
Kernel/platform-multicomp09/config.h:#undef   DEVICE_IDE        /* enable if IDE interface present */
Kernel/platform-multicomp09/README:* BUG other platforms have mixture of CONFIG_IDE and DEVICE_IDE and CONFIG_SDC and DEVICE_SD
```

Observe: DEVICE_IDE is never *used*. Therefore, it can simply be removed. What this highlights is that there is no direct linkage between IDE support and the kernel itself - any linkage is established by platform-specific files and the platform-specific Makefile (ie, by pulling in IDE support code from Kernel/dev).
```
$ grep -R CONFIG_IDE *
Kernel/platform/config.h:#undef   CONFIG_IDE
Kernel/platform-zx128/config.h:#define CONFIG_IDE
Kernel/platform-zx128/devices.c:#ifdef CONFIG_IDE
Kernel/platform-zx128/devices.c:#ifdef CONFIG_IDE
Kernel/platform-plus3/config.h:#define CONFIG_IDE
Kernel/platform-plus3/devices.c:#ifdef CONFIG_IDE
Kernel/platform-plus3/devices.c:#ifdef CONFIG_IDE
Kernel/platform-multicomp09/config.h:#undef   CONFIG_IDE
Kernel/platform-multicomp09/devices.c:#ifdef CONFIG_IDE
Kernel/platform-multicomp09/devices.c:#ifdef CONFIG_IDE
```
This reveals the true use for CONFIG_IDE - for a platform that supports IDE, CONFIG_IDE can used as a way of building the platform with and without IDE.

````
$ grep -R DRIVE_COUNT *
Kernel/dev/devsd_discard.c:    for(sd_drive=0; sd_drive<SD_DRIVE_COUNT; sd_drive++)
Kernel/dev/devide.h:#ifndef DRIVE_COUNT
Kernel/dev/devide.h:#define DRIVE_COUNT 2           /* at most 16 drives without adjusting DRIVE_NR_MASK */
Kernel/dev/devsd.h:   Define SD_DRIVE_COUNT to the number of SD cards your hardware
Kernel/dev/devide_discard.c:    for(d=0; d < DRIVE_COUNT; d++)
Kernel/platform-n8vem-mark4/config.h:#define SD_DRIVE_COUNT 1
Kernel/platform-zeta-v2/config.h:        #define SD_DRIVE_COUNT 1
Kernel/platform-msp430fr5969/config.h:#define SD_DRIVE_COUNT 1
Kernel/platform-msx2/config.h:#define SD_DRIVE_COUNT 1
Kernel/platform/config.h:#define SD_DRIVE_COUNT 1  /* 1 drive */
Kernel/platform-dragon-nx32/config.h:#define SD_DRIVE_COUNT     1       /* Could be higher with multiple CS used */
Kernel/platform-socz80/config.h:#define SD_DRIVE_COUNT 1
Kernel/platform-multicomp09/config.h:#define SD_DRIVE_COUNT 1  /* 1 drive */
````

Therefore, I propose:

- change comment in Kernel/dev/devide.h
- add description of CONFIG_IDE in PORTING
- change DEVICE_IDE to CONFIG_IDE in the config.h files with a short boilerplate comment
- change DRIVE_COUNT to IDE_DRIVE_COUNT
- add description of IDE_DRIVE_COUNT in PORTING

-----------------------------
````
$ grep -R DEVICE_SD *
Kernel/dev/devsd.h:   Define DEVICE_SD if SD hardware is present on your platform.
Kernel/platform-n8vem-mark4/config.h:#define DEVICE_SD
Kernel/platform-zeta-v2/config.h:       #define DEVICE_SD
Kernel/platform-msx2/config.h:#define DEVICE_SD
Kernel/platform-msx2/devmegasd.c:#ifdef DEVICE_SD
Kernel/platform-multicomp09/config.h:#define DEVICE_SD
Kernel/platform-multicomp09/config.h:#define DEVICE_SDC
Kernel/platform-multicomp09/README:* BUG other platforms have mixture of CONFIG_IDE and DEVICE_IDE and CONFIG_SDC and DEVICE_SD
````
As for DEVICE_IDE above, DEVICE_SD is only used in the comment of devsd.h and in one platform file (where it controls whether SD-related code is compiled). As for the IDE driver, there is no direct linkage between SD support and the kernel itself - any linkage is established by platform-specific files and the platform-specific Makefile (ie, by pulling in SD support code from Kernel/dev).

````
$ grep -R CONFIG_SD *
Kernel/platform-multicomp09/config.h:#define CONFIG_SD
Kernel/platform-multicomp09/devices.c:#ifdef CONFIG_SD
Kernel/platform-multicomp09/devices.c:#ifdef CONFIG_SD
````
Well, I can't use this to claim any precedent. Nevertheless, I will make a _consistency_ argument
in favour of changing DEVICE_SD to CONFIG_SD throughout.

````
$ grep -R CONFIG_SDC *
$ grep -R CONFIG_SDC *
Kernel/platform-coco3/devices.c:#ifdef CONFIG_SDC
Kernel/platform-multicomp09/config.h:#define CONFIG_SDC
````
Although both coco3 and multicomp09 use platform-specific drivers, rather than the drivers in Kernel/dev, the fact that the platform Makefile establishes what files are compiled means that there is no conflict -- nothing to prevent coco3 from using the define CONFIG_SD. Tormod suggested using CONFIG_COCO3SDC but I'd propose simply using CONFIG_SD

Therefore, I propose:

- change comment in Kernel/dev/devsd.h
- add description of CONFIG_SD in PORTING
- change DEVICE_SD to CONFIG_SD in the config.h files with a short boilerplate comment
- (SD_DRIVE_COUNT is already in use)
- add description of SD_DRIVE_COUNT in PORTING

tormodvolden, 2016-05-20:
No, you misunderstood what I wrote. CONFIG_SDC (on coco3) is not about a generic SD card interface, it is for Darren Atkinson's "CoCoSDC" cartridge which emulates a floppy controller and uses an SD card. So I suggested renaming this config to CONFIG_COCOSDC where "COCOSDC" reflects Darren's product name and not directly any "coco" machine or "SD" card interface.

tormodvolden, 2016-05-20:
One could even imagine someone using both the CoCoSDC and a more generic SD interface (like the spinx512 board) on the same platform in which case it would make sense to have two different CONFIG's. Other than that I agree to your suggestions.

Is there some historical reason or convention for using both CONFIG_* and DEVICE_* ?

nealcrook, 2016-05-22:
Thanks for the explanation Tormod; makes sense to me now. I have adopted your suggestion. Regarding CONFIG_* and DEVICE_* my grepping of the code-base shows no other usage of DEVICE whereas there is (for example) CONFIG_FLOPPY. Combined with the observation that these defines don't affect the common kernel code at all, the CONFIG_ prefix seems more consistent.

nealcrook, 2016-05-22:
Pull request Janitorial #347 includes changes for this issue.

closed 2016-05-22 by EtchedPixels