EtchedPixels/FUZIX #346: libraries build fail in clean workspace - sys/userstructs.h: No such file or directory

nealcrook, 2016-05-19:
From a clean git checkout, build of libraries fails:

````
make -C Library tools/syscall_6809
make -C Library/libs -f Makefile.6809 TARGET=multicomp09

(snip snip)

make[1]: Leaving directory `/home/crook/Desktop/Public/_neal/mc6809_stuff/myFUZIXMaster/Library/libs/fuzix6809'
cat fuzix6809/syslib.l | tr " " "\\n" | sed -e "s/^/fuzix6809\//" >syscall.l
m6809-unknown-as -o setjmp_6809.o setjmp_6809.s
m6809-unknown-as -o ashlsi3_6809.o ashlsi3_6809.s
m6809-unknown-gcc -c -Os -I../include -I../include/6502 __argv.c
m6809-unknown-gcc -c -Os -I../include -I../include/6502 abort.c
In file included from ../include/unistd.h:6:0,
                 from abort.c:4:
../include/syscalls.h:13:29: fatal error: sys/userstructs.h: No such file or directory
compilation terminated.
make: *** [abort.o] Error 1
make: Leaving directory `/home/crook/Desktop/Public/_neal/mc6809_stuff/myFUZIXMaster/Library/libs'
````

work-around is to copy it from elsewhere:

$ cp Kernel/include/userstructs.h Library/include/sys

I'm not sure what the ideologically sound fix would be.

Neal.

EtchedPixels, 2016-05-19:
I'm still not quite sure - there's a make install that tries to do the right thing but possibly all the Library C files should just depend upon the Library/include/sys/userstruct.h with a rule to create it by copying perhaps ?

tormodvolden, 2016-05-19:
If this is on 6809, didn't I fix it here? https://github.com/EtchedPixels/FUZIX/commit/f37a754e2252fabc537d397953789a08064fcc60

nealcrook, 2016-05-19:
Tormod - apologies; I committed the crime of reporting a bug without verifying that it still existed in latest. I agree that this now works correctly for me; closing.

closed 2016-05-19 by nealcrook