EtchedPixels/FUZIX #358: Platform review for Thomson machines

pulkomandy, 2016-06-03:
This page: https://github.com/EtchedPixels/FUZIX/wiki/Platform-Review-Notes-(8bit) is a bit inaccurate about Thomson machines (last entry)

- Name is Thomson, not Thompson
- There are two ranges "MO" and "TO", with similar hardware but different memory maps. TO is slightly more flexible.
- Both 0000-3FFF and A000-DFFF can be mapped to any of the 32 RAM pages
- The video framebuffer can be mapped in the banked space (or is there a requirement in FUZIX to have apps and framebuffer mapped at the same time?)
- This leaves 4000-9FFF for the kernel using RAM only. The result is a 24K/32K mapping, so the kernel would still need banking using the 8K window at 4000-6000 (there are two 8K pages there).

The main problem with this machines, however, is the inability to map RAM at FFFx where the interrupt vectors live. This means all interrupts have to run through the monitor ROM code before reaching anything else (unless you replace the system ROMs inside the machine). It could be ok if you only need a 1Hz interrupt (there is a programmable timer), but if you need higher frequency, running through the monitor may make things too slow.

Memory map details: http://pulkomandy.tk/wiki/doku.php?id=documentations:devices:io.devices

This probably does not open for an easy FUZIX port to these machines, but at least it is technically correct.

tormodvolden, 2016-06-03:
- Apps and framebuffer need not be mapped at the same time
- An interrupt at ~10Hz for the scheduler would have been very useful
- For correcting the wiki, I /think/ you can git clone it and send patches
- Thanks for the link to the Thomson info! Are you among the people who maintained a 6809 gcc before?

pulkomandy, 2016-06-04:
I'm (unfortunately) only an user of gcc-6809 so far. But I don't think I have hit enough problems with it to start hacking on it.

I think this moves Thomson TO8 to the "silly" category, rather than impossible. I'll see if I manage to clone and edit the wiki then.

pulkomandy, 2016-06-04:
Ok, here is a patch for the wiki:
http://pulkomandy.tk/drop/0001-Fix-some-data-about-Thomson-machines.patch

tormodvolden, 2016-06-04:
It is this site http://nostalgies.thomsonistes.org/index.html that I had seen once and gcc 6809 was mentioned. And now I discovered OS9 on MO5 http://os9.forler.ch/ very cool :)

pulkomandy, 2016-06-04:
Yes, from there you can get the initial gcc-6809 port based on gcc 2.95.3. I suggest staying with the current 4.x based branch from lwtools, however :)

More french info about OS/9 for MO5 there: http://forum.system-cfg.com/viewtopic.php?f=25&t=6139 (including some interesting hardware to further extend the RAM and ROM, should that ever be needed...)

tormodvolden, 2016-06-04:
Great to see that there is recent activity on that front! Things like the ROMDisk-NG beg for a FUZIX port :)

The two separated 16K windows could work if user programs over 16K are linked into two segments, definitely a hassle but not a deal-breaker. Sometimes .text in one and .data in the other is all that is needed. Then .text can be on a flash ROM bank for the most frequent applications so no loading is needed... :) Other than that the bankfixed.c model should be fine, just mapping the two banks in tandem. Small modifications will be needed for loading into (and clearing) the two separated banks. BTW, all essential programs are under 16KB text, but sh and ls need a diet.

The wiki text says the kernel would be limited to 24K. I don't understand this because there are the two 16K banks that both can be used for kernel, plus the 8KB for .common at 4000-5FFF.

Argh... must stop searching eBay for Thomson gear...

pulkomandy, 2016-06-04:
I'm not very familiar with the FUZIX layout yet. Is it possible to have the kernel in banked space, and only .common in central, always-mapped RAM? In that case, then yes, the 24K limitation is only for .common.

This would give the following scheme:
Bank 1: always mapped at 6000-9FFF, for .common
Bank 0: always mapped at 4000-5FFF as two 8K pages, for .common if 16K is not enough, or for other things (pieces of kernel? stack? video RAM as in the default config?)
Bank 2: video RAM, if we use bank 0 for something else

Banks 3+4 or ROM: kernel

Banks 4 or 5 to 31: for running apps.

I don't know if the lwtools linker is smart enough to split the .text section in two parts for the biggest programs.

I suggest trying https://www.leboncoin.fr/annonces/offres/midi_pyrenees/occasions/?th=1&q=thomson%20to8 instead of ebay, you may get more reasonable price. If the people there don't want to accept bank payments or shipping outside of France, I can buy from them and forward the shipping to you in exchange for some help with getting FUZIX to run ;)

EtchedPixels, 2016-06-04:
The 6809 gcc and linker don't know how to do general banked binaries. For Z80 I modified the sdcc toolchain to do linker/compiler driven banking but for gcc it looks kind of hard!

tormodvolden, 2016-06-04:
Yes, only .common (typically less than 4KB) needs to be mapped in all the time. (Well, it doesn't have to be - the "dragon" port tries without).

lwlink cannot split sections automatically. But larger binaries are usually made from several .c files and it is relatively easy to link them into sepa

tormodvolden, 2016-06-04:
Oh, and lwlink can be hacked too :)

Thanks for the Leboncoin link. I am serious about needing to resist though! Famous last words.

pulkomandy, 2016-06-04:
For the video, you can use any of the first 4 pages of RAM, that means either:
- The one that is always mapped at 4000-5FFF as two 8K pages
- The one that is always mapped at 6000-9FFF, direct access to the full 16K
- Or the two others that are only accessible in the banked spaces

The VDU bypasses the CPU banking scheme and always uses the RAM page allocated to it. It can do so even when the page isn't visible to the CPU at all.

For banking with gcc-6809, there is (or was) at least some support using a "far call" system: https://groups.google.com/forum/#!msg/embeddednewbies/iRQ3mHYlslA/2G84sWvv4u8J
This is useable only for storing code in banks, and needs an helper routine to do the bank switching. And we need to check if it still works in current versions. Would this be enough, or are there more needs? (banked data?)

tormodvolden, 2016-06-04:
I don't know whether "far call" still works, but it would give the same extra work as my suggested approach: You would still need to sort the different .c files between different code pages (instead of different text sections). On the other hand it would allow keeping all text of larger program in one window so the other window is free for data. And the more flexible bank16.c model would allow smaller programs where text+data fits in one 16KB bank, to only consume one 16KB bank, compared to the 32KB tandem-banks - not sure how important that is since you still would have around 13 tandem banks.

tormodvolden, 2016-06-04:
lwlink could need a way to specify gaps in the layout. I have a two-line patch that does it for a hard-coded gap, so this shouldn't be too difficult. (It would also need to pad the bottom part of the binary to allow it to load correctly.) With this we don't need to manually split the larger user programs. Only a few tweaks in _execve() are needed.

EtchedPixels, 2016-06-04:
> The VDU bypasses the CPU banking scheme and always uses the RAM page
> allocated to it. It can do so even when the page isn't visible to the
> CPU at all.

So if you hide the video and map as much RAM as you can at once how
much can be mapped ?
> For banking with gcc-6809, there is (or was) at least some support
> using a "far call" system:
> https://groups.google.com/forum/#!msg/embeddednewbies/iRQ3mHYlslA/2G84sWvv4u8J
> This is useable only for [...]

In the gcc 6809 I've seen it also only works for functions that pass
their arguments in registers - which isn't actually that useful for
general purpose work.

> works in current versions. Would this be enough, or are there more
> needs? (banked data?)

If the compiler banking limit was fixed it ought to be sufficient. What
I do on the ZX Spectrum 128K which has a Z80 with ROM 0-3FFF RAM
4000-BFFF and banked RAM at C000-FFFF only is to build the kernel in
banks (3 x 16K in the Z80 case - one also holding fonts/video + space),
and the kernel data and common live in the 4000-7FFF space. C000-FFFF
holds one bank of any program running (or the kernel bank) and
8000-BFFF is updated by exchanging it with one of the other banks. It
makes task switching expensive but it turns out to be acceptable for a
single user Unixlike box. We pull a similar trick on MSX1 with the
Brazilian memory extenders.

The compiler changes I did for SDCC on Z80 were fairly simple. The
compiler generates

              push af
              call func
              pop af

instead a simple call. It's also adjusted so that it expects arguments
2 bytes further up the stack.

The linker replaces every push af/call/pop af between banks with a
call to the bank switching helper followed by a word holding the
address wanted. It also fixes up any function calls via arrays/tables
using stub functions.

pulkomandy, 2016-06-04:
On Sat, Jun 04, 2016 at 10:50:44AM -0700, EtchedPixels wrote:
> So if you hide the video and map as much RAM as you can at once how
> much can be mapped ?

56K: 2x16K bank + 24K fixed

* 0000-3FFF: 16K bank RAM
* 4000-5FFF: 8K fixed (2x8K pages available here, but let's keep things simple)
* 6000-9FFF: 16K fixed
* A000-DFFF: 16K bank RAM
* E000-FFFF: ROM and IO space

> > 
> > For banking with gcc-6809, there is (or was) at least some support
> > using a "far call" system.
> 
> In the gcc 6809 I've seen it also only works for functions that pass
> their arguments in registers - which isn't actually that useful for
> general purpose work.

If I read the document correctly (and if the implementation matches),
this is because of the bank switching code itself saving things to the
stack. The compiler itself has no problem allowing functions using the
stack for parameter passing to use a far call. The "hard" part is writing
the bank switching code so it saves the previous bank elsewhere then the
stack. If it needs to be reentrant (code in a bank calling another one
and then back), a separate "stack" must be used (I think it can be just
a pointer + a buffer large enough for a given number of nested calls),
otherwise a single byte static variable is enough.

tormodvolden, 2016-06-04:
With 32KB banked (for user or kernel) and 24KB fixed (for common and kernel) you are pretty much exactly matching what we run on dragon-nx32, where 6KB is "wasted" for video, and kernel text is around 45KB. So I'd say just forget about the compiler banking for now, if you just want to get started. You'll need a simple bootloader to load the kernel, some routines to get characters on the screen and read the keyboard, a driver to read from disk/floppy, and the routines for switching maps, and off you go, the rest is pretty much taken care of. Just sign up to the mailing list, we'll be delighted to help.

EtchedPixels, 2016-06-05:
The 2 x 8K looks very useful as the kernel + common should fit fine in 2 x 16K banks + the fixed 16K (with room to spare), so without trying you'd get 2 x 24K user banks without any split + the 16K/16K splits.

The separate stack is tricky. That was my first plan on Z80 but it got real messy because of things like task switching. It's probably a lot lot easier to do right on 6809 as you can manipulate the stack flexibly.

EtchedPixels, 2016-06-18:
Wiki updated

closed 2016-06-18 by EtchedPixels