EtchedPixels/FUZIX #359: memory error induced by "ls"

nealcrook, 2016-06-12:
the script add-extras creates and populates /usr/bin
when attempting to "ls /usr/bin" the system hangs. Seen on platform multicomp09 in emulation and real hardware. Confirmed on platform coco3 by Brett.

$ ls /usr/bin
__snip snip__
wall
wump
mW __hangs__

"wump" is the last file (alphabetically). The text mW does not relate to any file name.

Here are some clues:

1/ simply creating a large number of files in a directory is not enough to reproduce this problem (well, my naive attempt did not fail)

2/ ls -f /usr/bin also fails (invalid opcode for me!) which is odd if this is a resource issue,  as not sorting the output should be less resource-heavy

3/ cd /usr/bin followed by ls works successfully

4/ I cut down the number of files in the ucp-extras-script.txt (by about 8 files) and get to this:
ls /usr/bin
__snip snip__
units
wall
wump
m]
4: out of memory
wump
backgammon

note that wump appears twice and that backgammon should not have appeared (it had already been listed in its alphabetic position). The number "4:" is sometimes 3: or 5: 

For the avoidance of doubt, note that add-extras does run fsck on the disk image.

Neal.

EtchedPixels, 2016-06-16:
I've seen some similar oddities. Not sure if its a bug in our malloc implementation (which I'd like to replace with a better and smaller one anyway) stdio, or ls. At some point ls wants a rewrite 8)

The 4: out of memory is from the kernel so it'll vary by process id. What looks odd to me is that you see the 4: out of memory and then more lines of output. I'd have expected the sbrk failure to have caused malloc to fail and us to exit. All the checks in ls seem correct so my first guess is malloc/free/realloc bugs somewhere.

I'll merge the replacement malloc/free/realloc soon and see if that helps

EtchedPixels, 2016-06-17:
Seems to be ok with the new malloc. Please build a new ls with the updated libc and check if it's now ok

nealcrook, 2016-06-18:
new libraries tree would not build for me on 6809.

First problem was here:

make[1]: Leaving directory `/home/crook/Desktop/Public/_neal/mc6809_stuff/myFUZIX/Library/libs/fuzix6809'
cat fuzix6809/syslib.l | tr " " "\\n" | sed -e "s/^/fuzix6809\//" >syscall.l
m6809-unknown-gcc -c -Os -I../include -I../include/6502 asctime.c
asctime.c:8:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
make: *** [asctime.o] Error 1

Here's asctime.c:
     8  static void hit __P((char *, int));
     9
    10  static void hit(char *buf, int val)
    11  {
    12          *buf = '0' + val % 10;
    13  }

I just commented out line 8.

Next problem:

m6809-unknown-gcc -c -Os -I../include -I../include/6502 free.c
free.c: In function 'free':
free.c:18:20: warning: initialization makes pointer from integer without a cast [enabled by default]
free.c:24:11: error: '__mfreeptr' undeclared (first use in this function)
free.c:24:11: note: each undeclared identifier is reported only once for each function it appears in
free.c:24:41: error: dereferencing pointer to incomplete type
free.c:24:55: error: dereferencing pointer to incomplete type
free.c:25:13: error: dereferencing pointer to incomplete type
free.c:25:40: error: dereferencing pointer to incomplete type
free.c:28:13: error: dereferencing pointer to incomplete type
free.c:28:24: error: dereferencing pointer to incomplete type
free.c:29:5: error: dereferencing pointer to incomplete type
free.c:29:16: error: dereferencing pointer to incomplete type
free.c:30:5: error: dereferencing pointer to incomplete type
free.c:30:15: error: dereferencing pointer to incomplete type
free.c:32:5: error: dereferencing pointer to incomplete type
free.c:32:15: error: dereferencing pointer to incomplete type
free.c:34:11: error: dereferencing pointer to incomplete type
free.c:35:4: error: dereferencing pointer to incomplete type
free.c:35:16: error: dereferencing pointer to incomplete type
free.c:36:4: error: dereferencing pointer to incomplete type
free.c:36:15: error: dereferencing pointer to incomplete type
free.c:38:4: error: dereferencing pointer to incomplete type
make: *** [free.o] Error 1

at which point I got stuck...

EtchedPixels, 2016-06-18:
Ah forgot to commit the malloc.h so it found the include dir one

EtchedPixels, 2016-06-18:
Pushed

nealcrook, 2016-06-18:
Thanks, that improved things. A couple more things were broken for the 6809 library build. I just pushed a fix to you for those. With those done, the library builds successfully and I confirm (for a small amount of testing) that the "ls" problem is gone. Thanks.

closed 2016-06-18 by nealcrook