EtchedPixels/FUZIX #367: new malloc not failing on overflow

beretta42, 2016-06-24:
from: ld09 while scanning memory for biggest memory it can alloc:
https://github.com/EtchedPixels/FUZIX/blob/master/Applications/ld09/table.c#L27

calls into malloc and malloc in-turn calls into brkmore() to actually brk/sbrk the memory:

https://github.com/EtchedPixels/FUZIX/blob/master/Library/libs/malloc.c#L36

But the overflow test on line 36 doesn't seem to catch all the cases.  The following is a screen dump of ld09 starting up, with gratuitous printfs thrown in (code, below...)  Please note that all the number are in hex, and seem to jibe. nb's are 4 bytes long so the first iteration from the following goes something like this:

p = 8b16 = sbrk
nb = 8a00 / 4 = 2281
therefore  p + nb = 8b16 + ( 2281 * 4 ) = 1151a 

```
sbrk = 8b16, nb = 2281, + = 151a
p + nb overflow
i=8a00, malloc=0
sbrk = 8b16, nb = 2201, + = 131a
p + nb overflow
i=8800, malloc=0
sbrk = 8b16, nb = 2181, + = 111a
p + nb overflow
i=8600, malloc=0
sbrk = 8b16, nb = 2101, + = f1a
p + nb overflow
i=8400, malloc=0
sbrk = 8b16, nb = 2081, + = d1a
p + nb overflow
i=8200, malloc=0
sbrk = 8b16, nb = 2001, + = b1a      <---- This and above fails like it should....
p + nb overflow
i=8000, malloc=0
sbrk = 8b16, nb = 1f81, + = 91a        <---- This brkmore call does not ????
i=7e00, malloc=8b1a <--- this makes sense considering a 4 byte memh header
ld09: out of memory while processing input file crt0.o
```

Here is my modded brkmore() for debugging:

```
static struct memh *brkmore(size_t nb)
{
	struct memh *p;

	if (nb < BRKSIZE)
		nb = BRKSIZE;

	/* Here be dragons: sbrk takes a signed value. We can however in C malloc
	   a size_t: We also assume nobody else misaligns the brk boundary, we won't
	   fix it if so - maybe we should ? */

	/* Current end of memory */
	p = sbrk(0);
	fprintf(stderr,"sbrk = %x, nb = %x, + = %x\n", p, nb, p+nb );
	if (p == (struct memh *) -1)
		return NULL;
	/* Overflow catch */
	if (p + nb < p){
		fprintf(stderr,"p + nb overflow\n");
		return NULL;
	}
	/* Move our break point. Using brk this way avoids the sign problems */
	if (brk(p + nb)){
		fprintf(stderr,"err on brk()\n");
		return NULL;
	}
	/* Fake it as a used block and free it into the free list */
	p->size = nb;
	free(p + 1);
	return __mfreeptr;
}
```

Here is my modded syminit (in table.c of ld09 )  for debugging:

```
PUBLIC void syminit(void)
{
    unsigned i;

    for (i = sizeof(int) <= 2 ? 0xE000 : (unsigned) 0x38000;
	 i != 0; i -= 512){
	    tableptr = malloc(i);
	    fprintf(stderr,"i=%x, malloc=%x\n", i, tableptr );
	if ( tableptr != NUL_PTR)
	    break;
    }
    if (tableptr == NUL_PTR)
	outofmemory();
    tableend = tableptr + i;
    for (i = 0; i < HASHTABSIZE; i++)
	hashtab[i] = NUL_PTR;

    mainavail = tableend - tableptr;
    usedtop = 0;
}
```

Is this some sort of pointer math fail ?  (gcc again?) notice the fail happens from i=8000 to i=7e00 or is this a red herring?

EtchedPixels, 2016-06-24:
That smells to me of compiler bug

Does it get it right if you make the if instead

if ((uint16_t)p + 4 * nb < (uint16_t)p)

?

beretta42, 2016-06-24:
Yes, same exact failure.  I tried casting everything to uint32_t also and testing for >= 0x10000  same results.  Suspecting a gcc failure, I tried compiling malloc.c with "-O0" - Still didn't work.

beretta42, 2016-06-26:
hmmm.. scrub my last comment.  I may have failed to recompile ld09 with modded malloc.  The gcc -O0 code looks impeccable - and for a change is sweet and simple.  I'll try again with -O0, and Alan's suggestion.  I'm betting it'll work.

beretta42, 2016-06-26:
Yup... confirmed: I forgot to remake ld09.  OK... -O0 makes it work.  So does Alan's casting suggestion.  BTW: ld09's aggressive scanning of memory works but the Kernel spits out a bunch of messages on console about being out of memory.  How should I proceed?

EtchedPixels, 2016-06-28:
I'll push the uint16 version as that's valid and just as clear as the original.

For the out of memory case well you could do something like this (and perhaps we need to have
an __memfree in libc that also hides any platform specific funnies)

size_t memfree(void)
{
                unsigned int room;
                uint8_t *bottom = sbrk(0);

                if (bottom == (uint8_t *)-1)  { shit happened }

                room = &room - bottom;
                return room;
}

then grab something like memfree() - 512;

EtchedPixels, 2016-06-29:
Changed to use uint16_t

closed 2016-07-01 by beretta42