EtchedPixels/FUZIX #370: Doc: emulators

vandys, 2016-07-07:
(Basically an enhancement request for the Wiki.)

It might be a big win to document how to start with some pre-built disk images
and config files for the obvious emulators--simh and MESS come to mind.
At least a 6809 target (MESS coco3h maybe) and a Z-80 one.  Maybe just
floppy based, maybe hard drive images too if it didn't add too many variables.
(I'd be happy to host floppy and hard disk images if there's an issue with that
here on Github.)

Thanks!

nealcrook, 2016-07-07:
I think it's annoying that the WIKI is disconnected from the code, discouraging updates. I wonder whether it would work better to copy the WIKI files to the Documentation tree and then periodically refresh the WIKI from that tree. It doesn't allow easy markup editing but it would allow contributors to make merge requests with WIKI updates. Since Markup can be more-or-less as readable as plain text, having it available in the development tree should be a bonus.

beretta42, 2016-07-11:
@vandys I have some images posted:
https://sites.google.com/site/cocoboot2/fuzix
Sry so late

vandys, 2016-07-24:
I've added some notes to the Coco3 platform README.  Unless anybody comes up with
something else to be done, I'll call it good and request retirement of this bug.

closed 2016-07-24 by EtchedPixels