EtchedPixels/FUZIX #371: ucp: double indirect blocks broken

beretta42, 2016-07-07:
UCP makes the correct filesizes, but leaves all zeros for any double indirect blocks.  Transfering this file back to linux reveals again a correct size, but anything after the direct blocks (sector 18) is zeroed!

In Fuzix:
```
# ls -l test.bmp
-rw-rw-rw-   1 root     0          196730 Jul 07 19:24 test.bmp
# od -hx test.bmp
<snip snip>
00223a0 d2ff ddd0 d2ff ddd0 d1ff ddd0 d2ff ddd0
00223b0 d2ff ddd0 d2ff ded1 d3ff ded1 d2ff ded1
00223c0 d3ff dfd2 d3ff ded1 d3ff ded1 d2ff ded1
00223d0 d2ff dcd0 d1ff dccf d1ff ddd0 d1ff ded1
00223e0 d2ff ded1 d3ff ded1 d3ff ded1 d2ff ddd1
00223f0 d2ff ded1 d2ff ddd1 d4ff dcd0 d6ff dbd0
0022400 0000 0000 0000 0000 0000 0000 0000 0000
*
0030000 00ff 0000 00ff 0000 00ff 0000 00ff 0000
*
0030070 00ff 0000 00ff 0000 00ff
003007a
```

If my math is logic is correct 0x22400 should be the first double indirect block?  
(no of direct + indirect block / 2 bytes per blkno ) * bytes per block = 0x22400
(18+512/2)*512=0x22400.

And the original file in Linux:

```
beretta@lappy:~/C/FUZIX/Applications/fview$ od -tx1 test5.bmp -Ax1
<snip snip>
02d420 bb ff c5 ba bb ff c5 ba bb ff c5 bb bb ff c3 b9
02d430 b9 ff c2 b8 b8 ff c0 b6 b5 ff be b5 b2 ff be b5
02d440 b2 ff c3 b5 b9 ff c5 b6 bb ff c3 b7 bb ff c2 b8
02d450 bb ff c3 b8 bb ff c2 b7 ba ff c1 b7 ba ff bf b6
02d460 b9 ff bf b6 b9 ff bd b6 b9 ff bc b5 b8 ff bb b4
02d470 b7 ff b9 b2 b5 ff b7 b3 b5 ff 00 00 00 ff 00 00
02d480 00 ff 00 00 00 ff 00 00 00 ff 00 00 00 ff 00 00
*
030070 00 ff 00 00 00 ff 00 00 00 ff
03007a
```

Note: My native tar seems to make bigger files than 1M with no problems, so I'm assuming this is a UCP thing and not the actual fuzix kernel (I think)

EtchedPixels, 2016-07-23:
Try this 

[1.txt](https://github.com/EtchedPixels/FUZIX/files/379988/1.txt)

beretta42, 2016-07-24:
yup that worked

closed 2016-07-24 by beretta42