EtchedPixels/FUZIX #374: coco3 build of assembly source

vandys, 2016-07-10:
Starting to attempt actual builds of the coco3 target.  The build rules inflict m6809-unknown-as on the coco3.s source.  This is just a wrapper around as6809, which barfs on lots of the
coco3.s source code, starting with the "include" directives.

Other parts of the source are built with lwasm, and that actually does quite a bit better
on this source, but barfs with:

coco3.s(13) : ERROR : Undefined exported symbol
coco3.s:00013             .globl interrupt_handler

coco3.s(34) : ERROR : Undefined exported symbol
coco3.s:00034             .globl _ramsize

and more of the same.

So should I be wrestling to make coco3.s work with as6809?  Or should I be trying to
fix lwasm to deal with whatever way it's gotten confused on import versus export
symbols?  I'm using gcc6809 4.3.6, and the latest/greatest for lwasm from their
web site. 

(Oh, I should mention that compiling C source is working fine.  So more than a bit
of gcc6809 is built and installed OK in any case.)

pulkomandy, 2016-07-10:
You need to use gcc-6809 version 4.6, which is distributed as a patch inside lwtools archives:
http://lwtools.projects.l-w.ca/hg/index.cgi/file/a3711f5ac569/extra

This one provides an "as" that wraps around lwasm instead of as6809.

vandys, 2016-07-11:
Thanks.  With current lwtools, I've managed to build the tree.  I'm going to note build issues here, but put together some markups to platform-coco3/README and submit a pull request
and then ask Alan to close this bug.  Once I have confidence it's all working!

That said, labs() seems to have become a builtin, so the source tree's definitions causes
warnings and at least one error.

m6809-unknown/libgcc doesn't build, but
m6809-unknown/fpic/libgcc did.  I used its output, but since I haven't booted my
results, dunno if that was an actual Good Idea (tm).  It let the apps link, though.

The README doesn't address building Applications/V7/cmd/sh/, which is needed to
build the filesystem image.

I now have two images, both named fuzix.dsk--one in Kernel/platform-coco3,
the other in Standalone/filesystem-src.  I'm guessing one's a floppy image
from which I boot ("dos" command?).  The other I'm guessing is a drivewire
disk image?  Is it known whether mame will boot and run these?  Is attaching
and booting these intuitively obvious?

Thanks again!  --Andy

beretta42, 2016-07-11:
https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/platform-coco3/build

This is the script I use.  It should build the basics.  Its helpful to have a drivewire server.  I'm not at my desktop at the minute, but I see where the README is a bit skimpy on examples for emulators. And my device list it quite larger now.  Does mess do IDE? I mostly emulate with this:
http://www.coco4.com/vcc/download.shtml
Use the fuzixfs.dsk as IDE drive 0 and boot with this:
LOADM"BOOT":EXEC 'HDA 
Any help on updating the README would be awesome.

-brett

vandys, 2016-07-11:
Don't have any Windows here, so will have to see what I can do with mame (mess has
apparently been subsumed into it!).  Building a version with Drivewire and we'll see
how it goes.  So far as I know mess has always had VHD hard drive emulation, and
in newer versions Drivewire as well.  If you want to chase this off of the issue tracker
you can reach me by searching: contact Andy Valencia, and the VSTa one will do
the job.

tormodvolden, 2016-07-11:
BTW, for anyone on Debian or Ubuntu, the gcc6809 and lwtools packages here have everything ready to go:
https://launchpad.net/~tormodvolden/+archive/ubuntu/m6809

vandys, 2016-07-13:
Here's some proposed markups, let me know what you think.  I'll submit a pull
request once you all think they're OK.  Thanks!
[rdm.txt](https://github.com/EtchedPixels/FUZIX/files/360650/rdm.txt)

beretta42, 2016-07-13:
Looks great. And  I learned stuff about mame :)  Please PR.  Then I'll  give it a once over to add the newish devices, vcc emulation guidance, more links to other oss dw servers.  I need to add some man pages too.

vandys, 2016-07-15:
platform-coco3 README markups in place, thanks!

closed 2016-07-15 by vandys