EtchedPixels/FUZIX #381: tc[sg]etattr fails for user

vandys, 2016-07-23:
I start up, have a shell as root.  su to a user account, then try to use a char-at-a-time
program (less, in this case, but a simple test program reproduces this).  The TTY does
not enter cbreak mode.  tcsetattr() (and tcgetattr() before it) fail with EPERM.

ioctl() looks up the inode and applies its permissions, which the sub-shell user does
not have.  Thus, no changing of TTY mode.

My vague memory of this part of the TTY Gordian knot is that you're allowed to ioctl()
a TTY file if it's your controlling TTY.  (I'd be happy to code up whatever approach you
prefer.)

vandys, 2016-07-23:
[tap.txt](https://github.com/EtchedPixels/FUZIX/files/379260/tap.txt)

vandys, 2016-07-23:
OTOH, V7 let you use ioctl's if you could open the device.  We could just leave off the
writability check, and if needed defend truly dangerous ioctl's with the existing
IOCTL_SUPER flag.

EtchedPixels, 2016-07-23:
Fixed

closed 2016-07-23 by EtchedPixels