EtchedPixels/FUZIX #402: pipe should return 0 when no more data

dfffffff, 2016-08-28:
When the write end of a pipe has closed and no more data is available, 0 should be returned to the application instead of -1 and EWOULDBLOCK.
This is a serious issue that breaks the pipeline.

Step to reproduce the issue:
```
# cat /etc/motd | cat

Welcome to FUZIX.
-: Try again
#
```
```
# cat /etc/motd | tr . . | tr . .
#
```

I have a patch for this issue, but unfortunately it hide another issue.
I got a SIGCHLD when execing:
```
# cat /etc/motd | cat | cat

Welcome to FUZIX.
-: Interrupted system call
#
```

The half-baked patch:
```
diff --git a/Kernel/inode.c b/Kernel/inode.c
index 1c590db..7f78619 100644
--- a/Kernel/inode.c
+++ b/Kernel/inode.c
@@ -52,7 +52,10 @@ void readi(inoptr ino, uint8_t flag)
 		}
 		toread = udata.u_count = min(udata.u_count, ino->c_node.i_size);
 		if (toread == 0) {
-			udata.u_error = EWOULDBLOCK;
+			if (ino->c_refs == 1)
+				udata.u_count = 0;
+			else
+				udata.u_error = EWOULDBLOCK;
 			break;
 		}
 		goto loop;
```

EtchedPixels, 2016-08-28:
The refs thing isn't sufficient I think because we need to know if there are any other writers not if there are any other openers.

The signal one I have an idea about - I've not seen it on my box (timing I guess) but I think the right answer is probably to

make chksigs check udata.u_cursig != 0 and if so just return

then in psleep_wait_io() call chksigs() then check udata.u_cursig rather than peering wrongly into the pending flags. That basically makes chksigs() something you can call to check your signal state whenever you need to get a current status. In fact we could even make chksigs return udata.u_cursig so it's a more hidden and natural check.

The pipe one looks trickier, as we need to know if there are any writers. That's partly a tweak to doclose() to spot

                if (getmode(ino) == MODE_R(F_PIPE) && O_ACCMODE(of_tab[oftindex]))
                        /* DO SOMETHING FIXME */
                }

and partly to figure out what the DO SOMETHING FIXME is and how to spot it in readi

EtchedPixels, 2016-08-28:
Humm and for named pipes we need to handle the case it gets re-opened

EtchedPixels, 2016-08-28:
Ok messier than I would like but something like this perhaps ?

[pipe.txt](https://github.com/EtchedPixels/FUZIX/files/441410/pipe.txt)

EtchedPixels, 2016-08-28:
Test fixes to both pushed (untested as its midnight and I'm off to bed!)

dfffffff, 2016-08-29:
Tried your patch, great for the EWOULDBLOCK, however the SIGCHLD is still an issue.
Only the parent should receive a SIGCHLD not the siblings, I will check tomorrow if I can do something (I'm off to bed too!), I think the issue is located in process.c at signal_parent().

EtchedPixels, 2016-08-29:
I need to  take another look at the pipe one as well. I think it might actually be as memory efficient and lots faster to just add a uint8_t reader and writer count to the c_inode.

EtchedPixels, 2016-08-29:
This all now works with my tree - please double check if you still see a signal funny and if so trace back what cursigs() is up to

dfffffff, 2016-08-29:
Do you have the SIGCHLD issue on your platform?
Might be related to my implementation.

I added a signal handler to /bin/cat util.
I get this at _fork() syscall
```
# cat /etc/motd | cat | cat

fork: current 2, child 3

fork: current 2,
fork: current 4, child 5
 child 4

Welcome to FUZIX.
in pid 4, received signal 17
#
```
'current' should always be 2, pid of /bin/sh.
Looks like child 4 fork and create child 5
and child 5 get the wrong parent pid.

dfffffff, 2016-08-29:
Okay, the problem reside in sh, I have compiled the sh source for my host computer, and I have the same weird issue with the children.
So sh need to be fixed.

dfffffff, 2016-08-29:
Maybe we can close this ticket and open a new one for sh ?

EtchedPixels, 2016-08-29:
If you look how the Bourne shell builds pipelines it doesn't necessarily make all the processes direct children of the shell. Even then it shouldn't matter as in the normal case SIGCHLD with SIG_DFL is supposed to be ignored (so doesn't cause an interrupt of anything). If you install a handler then you will see the SIGCHLD for other bits of the pipeline I believe

If I read it rightly its forked as

2 shell
3 cat /etc/motd  parent 2
4 cat parent 2
5 cat parent 4

For all of the pipeline the SIGCHLD will be SIG_DFL so ignored unless the process asked for it specifically and you shouldn't get any interrupted read/writes (hence I want to know what chksigs is returning)

dfffffff, 2016-08-29:
Ok great, I have not tried your patch with no signal handler, everything seems to work now.

closed 2016-08-29 by EtchedPixels