EtchedPixels/FUZIX #415: uname 6809

beretta42, 2016-08-31:
gcc fail?  uname is supposed to follow sysinfo.  

https://github.com/EtchedPixels/FUZIX/blob/85c14812033e1e69a27e48b64876af73d98a8106/Kernel/makeversion

```
$ grep sysinfo ../fuzix.map ; grep uname_str ../fuzix.map 
Symbol: _sysinfo (../version.o) = 7DDA
Symbol: _uname_str (../version.o) = 7DC6
```

dfffffff, 2016-08-31:
I think under gcc you can make no assumption on the order the code or data will outputs...

EtchedPixels, 2016-08-31:
I'm surprised that breaks but yes it's not a valid C assumption. B was much nicer 8)

dfffffff, 2016-08-31:
could uname_str and uname_len be incorporated into struct sysinfoblk ?

EtchedPixels, 2016-08-31:
That gets ugly because the struct has variable length. Probably we just have to accept it is two copies

dfffffff, 2016-08-31:
But if we implement an uname_r() we get rid of the static buffer and only one copy is made

closed 2016-09-01 by EtchedPixels