EtchedPixels/FUZIX #448: Replace the 'new' build system with something that works

davidgiven, 2016-10-29:
So... that build system that I wrote, which nobody else has been able to make work, and is always broken... yeah. That was a bad idea. Sorry. It needs to die.

But I still think it needs replacing with something else, but this time, something that's not terrible. What?

The more I use make, the more I loathe it. It's just inadequate, and broken. For the ACK I recently put in a concerted effort to figure out how to do targets with multiple outputs in make. The answer is, it can't do this. You can just about force it do to multiple outputs in GNU Make, in certain specific circumstances, but it's very fragile.

I'm becoming increasingly convinced that any non-trivial project in make only has a build system which works by accident, particularly if you want to do parallel builds. I don't believe robust builds in make are *possible* because there are just too many horrible edge cases.

I think Fuzix is too complicated to build reliably with make --- the dependencies between kernel and library and apps and tools are too complex. What I was trying to do with that horrible set of make macros was to build everything in a single pass, so make could be aware of all the dependencies at once and get it all right. And, on the rare occasions when it actually worked, it did actually do it; it just wasn't maintainable.

But what's the alternative? Is there anything else which is widespread enough to switch to without adding exotic dependencies to the project? I've been using ninja for the ACK, and it's great, but it's not a build tool --- it's merely the backend to a build tool, and you need a generator script to produce the ninja script, and arrgh life's too short. cmake and scons appear to be the leading contenders, but they look pretty grim too.

At work I use bazel, which is amazing, and is the only non-terrible build tool I've ever seen, but I suspect an 80MB download (not including the JVM!) wouldn't go down well.

vandys, 2016-10-30:
I've used the build/make utility "jam" in past lives.  It's not the be-all and end-all, but it
certainly is a lot better than make, which still being lightweight.  I see it's in the standard
Debian repos.

pulkomandy, 2016-10-30:
I was going to suggest jam too. The official version from perforce is
unfortunately not getting any updates for a few years, but you can get
versions from Haiku, Boost or Freetype, each with their own changes.

If what you're after is something in the same spirit as make, but a
little more sane, it is a good choice. That means it will require some
work to write rules and everything for FUZIX.

If you want something more higher level, cmake is nice, but in the case
of FUZIX, I think close to none of the default rules there will be
useful for anything.

So yes, since you need to write the rules for driving the various
compilers and linkers anyway, Jam may be a good choice. Also, I can try
to provide some help as I contribute to Haiku, so I have some experience
with Jam.

-- 
Adrien.

EtchedPixels, 2016-10-30:
I would like to either sort out the Makefile stuff or figure out how to fix it. I do ideally want to find something that will run on small machines though - 8bit Fuzix may never be self hosting but 68000 Fuzix at least ought to be able to manage it.

dfffffff, 2016-10-30:
What about something like kbuild system, linux manage to do great things with makefiles.

erkinalp, 2016-10-30:
> 8bit Fuzix may never be self hosting

Nothing like DJGPP exist for 8 bits?

vandys, 2016-10-30:
>8bit Fuzix may never be self hosting

This has been an interest of mine since i first learned about Fuzix--I feel sheepish to not even
think of it when I suggested Jam.  Self hosting on small systems does argue for a basic make
program with carefully designed hierarchy of build directories.

davidgiven, 2016-10-30:
I can't find any decent Jam documentation --- even Perforce's website is kinda spare these days. Anyone got a reference?

I'm assuming that on a 68k system, GNU make will be too big, so we'd be using an older make instead. (BTW, I've got a 29-year-old make right here: https://github.com/davidgiven/ack/tree/default/util/make) That means we wouldn't get any of the hacks to support rules with multiple outputs that GNU make has. Fuzix doesn't use this as much as the ACK does, but IIRC it does use them. But, of course, any such system is likely not to support parallel builds, which means it's not as critical to get this stuff exactly right.

Did you have any specific make in mind?

(Plan9's mk is an 88kB shared binary on my development box.)

davidgiven, 2016-10-30:
For 8-bit development, it is actually feasible to take the build dependency graph, do a postorder sort on it, and spit out a giant shell script containing lines like this:

    test foo.c -nt foo.o && cc -o foo.o foo.c

...which the shell will (well, should) read and execute a line at a time, so keeping memory usage absolutely minimal.

But that would most likely require a custom build tool, and that way lies madness.

EtchedPixels, 2016-10-30:
I need to have a look how *BSD does it I think, especially older BSD.

drawkula, 2016-10-30:
> I need to have a look how *BSD does it I think, especially older BSD.

[LiteBSD](https://github.com/sergev/LiteBSD) and [RetroBSD](https://github.com/RetroBSD/retrobsd)?

vandys, 2016-10-31:
> [LiteBSD and RetroBSD?]

Those are very nice projects, albeit quite a bit larger than Fuzix.

I've been reading RetroBSD, which indeed uses a hierarchy of make invocations.
On Fuzix this should work OK, so long as the upper make's can be swapped out to free
memory for the current target compilation.

(I don't like the idea of a monster host machine generating a long linear build file.
That doesn't feel at all ike true self hosting.)

mkoloberdin, 2017-03-31:
Consider Shake: http://shakebuild.com/why

vandys, 2017-03-31:
"Consider Shake"

Ummm, how do you see Haskell ever appearing on, say, a 6809 or Z-80?

mkoloberdin, 2017-06-01:
@vandys , is non-cross-compilation even a goal for this project? Is it doable?

EtchedPixels, 2017-06-01:
I think for 68000 and above self hosting should be possible in theory - ditto for 8086 just about.

For the 8bit platforms it's going to be a very tight squeeze to get any C compiler, let alone one with enough optimization to build the kernel usefully.

At the moment the 68000 code isn't stable enough and the swap support isn't good enough to start building 'serious' tools but it is getting there bit by bit.

vandys, 2017-06-01:
Well, V7 Unix certainly did it, as does BSD 2.11.  I can't think why it categorically should be off the table, although in a 32 bit world (shifting to 64 bits), it's very easy to make choices which would make it impossible.

EtchedPixels, 2017-06-01:
V7 did it in 120K of RAM on a PDP/11 with a regular instruction set. 2.11BSD does it using even more RAM and banking.

On our Z80 platforms we have perhaps 48K of RAM and a processor whose instruction set is less regular than a medical constipation emergency. It's not just about RAM either - SDCC has to do some serious crunching to generate good Z80 code, it would also take weeks to compile with the needed levels of optimization.

I definitely want a native C compiler in the end but I don't think we should expect it to generate good enough code to be totally self hosting.

Right now btw you can run the CP/M Hitech C compiler on a big enough Z80 box running Fuzix by wrapping it with runcpm. Add fuzix C libraries and some tools to fix the link and you can sort of do self hosting. However the compiler output while superb for a CP/M self hosted compiler is crap compared with SDCC and it's "free" but not in the open source sense unless someone wants to disassemble and label and comment it all up nicely.

davidgiven, 2017-06-01:
Minix 1 and 2 had a C compiler that could recompile its own kernel in a 16-bit address space. Of course, it had way more actual *RAM* than that, so the compiler didn't have to share that address space with the rest of the system, and it used a split I+D model process so you got 64kB of code *and* 64kB of data, but it was still damned impressive. Both systems were (almost) completely self-hosting. I, uh, hacked together a crude distribution if anyone wants to play. https://github.com/davidgiven/minix2

...but 8086 code is way easier to compile C into than the Z80 and 6502, due to being much friendlier to stack frames and pointer addressing, and the compiler isn't amenable to porting (I've tried). I haven't found any figures on relative code density for compiled C on the different architectures; a quick search just showed up a thread by me asking about this, from comp.os.cpm, from 2004, which I shall take as a sign that I need to go to bed.

EtchedPixels, 2017-09-29:
Crude but functional top level Makefile and glue added

closed 2017-09-29 by EtchedPixels