EtchedPixels/FUZIX #449: net_native.c hangs on signal

beretta42, 2016-10-29:
When a process is kill()ed, net_native hangs the system by psleep()ing the task, calling chksig() forever until death.

chksigs()  (process.c)
doexit() (process.c) 
net_close()
netn_synchronous_event()
psleep() 
switchout()
chksigs()   <--- goto 10 repeat until system death.

I fixed via this:

```
diff --git a/Kernel/process.c b/Kernel/process.c
index 5bc3adb..f96f0b8 100644
--- a/Kernel/process.c
+++ b/Kernel/process.c
@@ -505,6 +505,7 @@ rescan:
 #ifdef DEBUG
                        kprintf("process terminated by signal %d\n", j);
 #endif
+                       udata.u_ptab->p_pending &= ~m;
                        doexit(dump_core(j));
                } else if (*svec != SIG_IGN) {
                        /* Arrange to call the user routine at return */
```

EtchedPixels, 2016-10-30:
Argh yes the core code assumes close() doesn't block, which is a bad assumption all round.

I don't think your fix is sufficient. The p_pending clear needs doing and can be done by just moving it out of the other bit of the if, but I think we also need to set p_held to 0xFFFFFFFF in doexit() so that any other signal arriving while we are sleeping for the exit can't cause a problem and recursively exit

Test patch pushed

beretta42, 2016-10-30:
Pushed fix seems to fix my limited problem case.   Killing a processes with open sockets doesn't crash my system anymore :)  I'll keep playing with killing tasks and report any problems.

closed 2016-10-30 by EtchedPixels