EtchedPixels/FUZIX #457: What is the Right Way to use /Build/tools/m6809-* ??

nealcrook, 2016-11-15:
Question for Tormod:

Recent changes to the m6809 makefiles have switched from the direct use of lwlink to the use of the m6809-unknown-ld wrapper, together with the use of the "--oformat" switch.

The released m6809-unknown-ld wrapper does not support the --oformat switch, only the "--format" switch (though the required effect is the same - this is a bug in the wrapper).

There are fixed wrappers in Build/tools and I understand that you have pushed these changes upstream to William Astle for inclusion in a future LWTOOLS release.

What is the recommended way to deploy these wrappers? Should I

1/ copy them over the top of the installed wrappers in /usr/bin
2/ change my PATH to add <wherever>/Build/tools
3/ some other cunning method I've not thought of?

thanks,

Neal.

beretta42, 2016-11-15:
> 2/ change my PATH to add /Build/tools

I threw /Build/tools in my PATH.

brett

dfffffff, 2016-11-15:
like brett said the best way is to add a PATH to /Build/tools

tormodvolden, 2016-11-15:
The Kernel/README.md recommends prepending Build/tools to your path. The idea is that these tools override any coming with stock LWTOOLS.

nealcrook, 2016-11-15:
thanks; it's unanimous and I have done as you suggest. <blush>If only I'd read README.md</blush>

closed 2016-11-15 by nealcrook