EtchedPixels/FUZIX #459: Missing vfscanf_m.c breaks Library/libs build

protoCall7, 2016-11-19:
When building Library/libs on Mac OS X with sdcc 3.6.0 #9615, make fails on a missing dependency for vfscanf_m.c.

```
fcc -O2 -c -DBUILD_LIBM scalbnf.c -o scalbnf.rel
fcc -O2 -c -DBUILD_LIBM scalbinf.c -o scalbinf.rel
fcc -O2 -c -DBUILD_LIBM sinf.c -o sinf.rel
fcc -O2 -c -DBUILD_LIBM sincosf.c -o sincosf.rel
fcc -O2 -c -DBUILD_LIBM sinhf.c -o sinhf.rel
fcc -O2 -c -DBUILD_LIBM sqrtf.c -o sqrtf.rel
fcc -O2 -c -DBUILD_LIBM tgammaf.c -o tgammaf.rel
make: *** No rule to make target `vfscanf_m.c', needed by `vfscanf_m.rel'.  Stop.
```

Searching the source tree shows no sign of the missing file, and the only reference to it that I'm able to find is in the Makefile itself.

EtchedPixels, 2016-11-19:
Fix pushed

protoCall7, 2016-11-19:
Build completes successfully now. Thank you!

closed 2016-11-19 by protoCall7