EtchedPixels/FUZIX #460: Applications/utils build fails on seltest.c

protoCall7, 2016-11-19:
This appears to be a similar issue to #459 
```
../../Library/tools/fcc   sed.rel -o sed
sed: 21388 bytes from 256
../../Library/tools/fcc  -c socktest.c
socktest.c:159: warning 84: 'auto' variable 'fd2' may be used before initialization
socktest.c:171: warning 84: 'auto' variable 'fd2' may be used before initialization
../../Library/tools/fcc   socktest.rel -o socktest
socktest: 7735 bytes from 256
make: *** No rule to make target `seltest.c', needed by `seltest.rel'.  Stop.
```

The only reference I found to this file is in the Makefile:

```$ grep -n seltest Makefile
104:SRCSBAD = sed.c socktest.c seltest.c
$ find . -name "seltest*"
$
```

protoCall7, 2016-11-20:
Removing the reference to seltest.c on line 104 of Applications/util/Makefile allows the build to complete successfully.

EtchedPixels, 2016-11-20:
oops. Added to the repository Thanks

closed 2016-11-20 by EtchedPixels