EtchedPixels/FUZIX #461: no #! support in the kernel

nealcrook, 2016-11-21:
I tried to write a shell script and it would not start. Am I correct in saying that support for #! is a function of the kernel and that the kernel does not currently support this...

````
ssh# ls -al test.sh
-rwxrwxrwx   1 root     root           29 Jan 01 00:00 test.sh
ssh# cat test.sh
#!/bin/sh
echo "Hello world"
ssh# ./test.sh
ssh: ./test.sh?
````

beretta42, 2016-11-21:
Correct.  "#" isnt supported either. ;)

EtchedPixels, 2016-11-21:
In older traditional unix #! is a feature of the shell (or C library). I believe the V7 shell (/bin/sh) will try and run something as a script if execve fails and it's executable. ssh doesn't have a clue about scripts 

The '#' comment feature in sh slightly postdates V7. It's something that 's probably worth importing from Heirloom /bin/sh (the later 32bit sh)

closed 2016-11-21 by EtchedPixels

nealcrook, 2016-11-22:
D'oh. Despite my example, I had actually tested this under sh as well as simple shell (aka ssh). The root cause of my problem was actually the unrecognised/unsupported "#". Without the "'#", the script is interpreted correctly - by the shell, as Alan says, and not by the kernel as it was in later Unix releases (as I would have known if I had read https://en.wikipedia.org/wiki/Shebang_(Unix)#History )