EtchedPixels/FUZIX #485: Nintendo Entertainment System Support

keito940, 2017-02-07:
...And Created NES FUZIX Cart(If you put bootstrap in ROM part, it will be able to go up to start.
The storage will be managed by the flash memory cart. (Battery backup is full of defects.)) And Keyboard Device And Mouse Device (Family Basic is Rare). 
...And Super NES(65816) & Genesis(Main:68k+Sound:Z80) Support And Create Genesis & Super NES Keyboard Device?

beretta42, 2017-02-07:
The nintendo platforms where pretty limited on RAM no?

atsidaev, 2017-02-07:
Cart can have onboard RAM. Using it the resulting memory map will be close to that of ZX Spectrum, where only 0x4000-0xFFFF block is available (will be 0x4020-0xFFFF in NES case).

EtchedPixels, 2017-02-07:
If you want support send patches.

closed 2017-03-24 by EtchedPixels