EtchedPixels/FUZIX #486: PDP11 T11 port?

wa2mze, 2017-02-11:
I'm a former DEC employee (my first job out of college) who cut his teeth on the LSI11 family.  I've got a T11 micro processor chip in the junk box and I'm thinking of building a machine around it.  I intend on adding a 74HCT612 memory mapper chip, so I'll be able to approach the emulation of the memory structure of the PDP11/40,45,70 models with 18, 22, or even 24 bit addressing, with eight 8K byte banks of memory, and maybe even "I" and "D" spaces (if the processor 'fetch' status correctly indicates all access via the PC).  No division of User, Kernel, and System priority though.  I'd like to hear from anybody interested in this, with any comments or advice on getting FUZIX running on it. (If I only had machine readable copies of the RT11 sources or binaries, or the Falcon board ODT  :-) )
Thanks!

erkinalp, 2017-02-11:
Direct competitor of UNIX from the future.

beretta42, 2017-02-11:
Very cool.  I would look forward to following you on your build.   I have a hunch you'll have no problem porting fuzix to such a machine.  I guess you'll need ti choose a memory model first.

EtchedPixels, 2017-02-12:
I did reply on the Google+ post - the core code builds with gcc for pdp-11 but I never got time to sit down with the hardware manuals and SIMH to write a loader and some device drivers.

For the memory model it's going to depend a lot on the nature of the system. Older 11's have an MMU where fault handling on instruction faults is near impossible so might need some gcc tweaks (stack probe instructions) to do "proper" virtual memory, but the later systems can do that right.

For a platform with a simple external mapper I'd guess you'd go with a chmem like model at least to start with and just use the 8K banking code (possibly with Brett's changes for a fixed common if you don't have separate memory spaces for user and supervisor).

The good thing is Fuzix doesn't really care too much, providing you've got enough memory, some way to switch banks and some I/O it'll cope with the rest 8)

closed 2017-11-07 by EtchedPixels