EtchedPixels/FUZIX #495: Implement Java (or C#) style programming language and Python

keito940, 2017-03-09:
Java And C#.
Java is a programming language with C ++ usability added to Objective-C ported Smalltalk-VM to C language.
In addition, C# is a programming language that contains elements of Java and C++ in Delphi which is a Pascal processing type used in Windows. Because there are things that Delphi former staff and Windows selling agency are making, it is getting familiarity with Windows and cross platform.
And the language Python is establishing its status as a split language.
So, can you port Python and Java or C#?
Even as it is the lightness of FUZIX as it is.

beretta42, 2017-03-13:
Sounds cool... I bet Alan would say: "sure! you send the patches in"   :)

EtchedPixels, 2017-03-24:
yep

closed 2017-03-24 by EtchedPixels