EtchedPixels/FUZIX #497: Clarify status with Floppy controllers: which, what, when needed etc

fgeorgatos, 2017-04-18:
we see that interfacing floppy controllers are on the demand; we may be able to help!
Can we be precise about what's needed?

I recall that in my early days, I was able to write sector reader code in z80 assembly for nec765a;
who knows, 30 years after the amstrad CPC, it might still be possible to revive that skill!
(anyone needing help near Cambridge, UK?)

Also, I had some esoteric experience with the z80 around R & I registers, 
while trying to hack the locking mechanisms of the time, which were quite elaborate:
http://www.cpcwiki.eu/forum/emulators/an-unpublished-tip-for-pyradev-users/
Basically, the trick to get there was to discover that the *Mustek* (clone) manuals were a better resource for some undocumented behaviour and how flags get modified; with that in hand:
* you can preserve the R register and the Interrupt Flag transparently, totally useful for debuggers and possibly some more; this may be handy if there is need for transparent context switching.

F.

EtchedPixels, 2017-04-18:
Ah the joys of annoying debuggers. I forget which game did it but one actually kept SP pointing into ROM for the entire game solely to break debug tools and multiface 80

There are now some nec 765 drivers in various ports. I gave up trying to unify them because they are so different. Still need to sort a few out like the Amstrad NC200.

CPC6128 is far down the to-do list and I'll probably do the very similar Spectrum +3 first.

fgeorgatos, 2017-04-20:
lol... SP pointing to ROM! 

reflecting on this, it made me realize that a locking mechanism for Z80, which would rely on both SP + R registers would be omnipowerful and quite difficult to handle. Combine that with a cylinder #42 floppy disk locking (popular back then, [1]) and you have the tzar-of-all-cpc-locking-tricks ; It may still be able to overcome by a smart-tracing debugger (detect `LD SP, *`) but, call that complicated to handle!

Which makes for a relevant question: is the floppy controllers' objective more about the practicalities of reading data in/out or, there is more to it, like the esoteric aspects/snags of each individual controller?!

[1] http://webcache.googleusercontent.com/search?q=cache:Iabr81bkv2kJ:cpctech.cpc-live.com/docs/upd765a/necfdc.htm+&cd=6&hl=en&ct=clnk&gl=uk

EtchedPixels, 2017-04-21:
esoterics of each platform - and also managing to hit the controller deadlines.

EtchedPixels, 2017-04-25:
and the other one is format commands...

fgeorgatos, 2017-04-26:
@EtchedPixels : as regards CPC series, the Nirvana tool had been a marvel for disk management:
http://www.cpcwiki.eu/index.php/Nirvana
http://cpcrulez.fr/applications_disc-nirvana__ACU.htm

I no longer recall why it was the tool of choice for formatting back then (via FDC nec765a); 
perhaps, it was because it was strong in recognising weird formats and would copy them across!
good thing is it was small and as such it is suitable for disassembling/cross-reference.

closed 2017-08-15 by EtchedPixels